mantidproject/mantid #30607: Investigate whether there is a python mesh viewer available

DannyHindson, 2021-02-08:
The single crystal instrument scientists at ISIS (WISH, SXD) have indicated that the lack of any screen to view sample mesh shapes in Mantid is a gap. They would like to view the shape to check that the shape and the orientation have been imported into Mantid correctly.

We have discussed the fact that using third party CAD tools is an option and may make more sense than Mantid trying to implement its own shape viewer. They have asked whether there is a python based viewer available.

DannyHindson, 2021-03-01:
I've had a related request through that could usefully be investigated at the same time. This one is about whether there is a way of viewing a CSG description of a shape somehow in Mantid or using an external tool. I'd like to look at whether there is a way of exporting the CSG description in a commonly used format to begin with. If the CSG format isn't widely supported we could even look at exporting the shape as a mesh,

There did used to be a way of viewing shapes in Mantid using OpenGL (similar to instrument viewer) but this feature never worked fully and it was disabled in MantidPlot and therefore never made it into Workbench. Speaking to Martyn, it doesn't seem like it's going to be v easy to re-enable this - especially now that we're moving away from OpenGL

RichardWaiteSTFC, 2021-03-01:
There seems to be a library PyMesh that can read .stl (and some other formats I hadn't heard of but not .csg) https://pymesh.readthedocs.io/en/latest/basic.html#loading-mesh
but it has limited plotting capability.
There is another library dedicated just to .stl that might be a better option which plots with matplotlib 
https://pypi.org/project/numpy-stl/
But no simple scripts - so it looks like we'd need to add a dependency to do this. 
I don't know whether there is a way to convert a .csg to a .stl (it looks like .csg is not a true mesh format)?

RichardWaiteSTFC, 2021-03-01:
Perhaps OpenSCAD can convert csg to stl and vice-versa
https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/STL_Import_and_Export

DannyHindson, 2021-03-01:
@RichardWaiteSTFC - thanks for looking into this. I hadn't realised there was a format for storing CSG descriptions of shapes. It seems like the .csg format might be OpenSCAD specific so we'll have to take a view on whether this is insufficiently general to use as an export format from Mantid. It might be better (but more work!) to convert the CSG description to a mesh for the purpose of viewing it in a 3rd party tool

DannyHindson, 2021-03-30:
Few latest thoughts from me on this ahead of our sprint planning:
- I like the look of https://pypi.org/project/numpy-stl/. The plotting in matplotlib seems to fit into the Mantid framework quite nicely. This would provide a nice solution for visualising meshes in Mantid. When I ran the example script I got a plot window with the Mantid toolbar and the toolbar buttons threw a few errors because it wasn't a standard workspace plot. So this would need tidying up
- the solution for viewing CSG objects is a bit less clear. My previous suggestion about converting CSG shapes to a mesh glosses over the effort involved in achieving this conversion. There is some code in the Geometry::GeometryHandler class that (I think) converts the CSG objects into a mesh for viewing but I think this has been written with OpenGL in mind. Might be possible to reuse this code to create a new class to plot in Matplotlib also?? Alternatively we could just write an algorithm to export the CSG shapes to OpenSCAD. I have a feeling that might be an easier first step

DanielMurphy22, 2021-04-23:
As the CSG and .3mf mesh formats aren't yet supported, I'm reopening!

DannyHindson, 2021-06-02:
This looks like where the existing "convert to mesh" code lives:
Framework\Geometry\src\Rendering\RenderingHelpersOpenGL.cpp

closed 2021-07-12 by martyngigg