mantidproject/mantid #30622: Wrong user propertis causes mantid to crash

AndreiSavici, 2021-02-08:
This is related to #30504 

### Expected behavior
Be able to start workbench
### Actual behavior
```
3y9@pc0122665:~/Mantid/Build$ bin/launch_mantidworkbench.sh 
/usr/bin/python3.8: No module named pip
bin/launch_mantidworkbench.sh: 39: [: Illegal number: 
/home/3y9/Mantid/Build/bin/site.py:26: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp  # Avoid import loop in Python 3
Traceback (most recent call last):
  File "/home/3y9/Mantid/Build/bin/workbench", line 11, in <module>
    load_entry_point('MantidWorkbench', 'gui_scripts', 'workbench')()
  File "/home/3y9/Mantid/mantid/qt/applications/workbench/workbench/app/main.py", line 46, in main
    start(options)
  File "/home/3y9/Mantid/mantid/qt/applications/workbench/workbench/app/main.py", line 50, in start
    from workbench.app.start import start
  File "/home/3y9/Mantid/mantid/qt/applications/workbench/workbench/app/start.py", line 29, in <module>
    from workbench.plugins.exception_handler import exception_logger  # noqa
  File "/home/3y9/Mantid/mantid/qt/applications/workbench/workbench/plugins/exception_handler/__init__.py", line 12, in <module>
    from mantidqt.dialogs.errorreports.report import CrashReportPage
  File "/home/3y9/Mantid/mantid/qt/python/mantidqt/dialogs/errorreports/report.py", line 30, in <module>
    class CrashReportPage(ErrorReportUIBase, ErrorReportUI):
  File "/home/3y9/Mantid/mantid/qt/python/mantidqt/dialogs/errorreports/report.py", line 34, in CrashReportPage
    interface_manager = InterfaceManager()
Exception: unknown
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/3y9/Mantid/mantid/qt/python/mantidqt/dialogs/errorreports/main.py", line 16, in <module>
    from mantidqt.dialogs.errorreports.report import CrashReportPage
  File "/home/3y9/Mantid/mantid/qt/python/mantidqt/dialogs/errorreports/report.py", line 30, in <module>
    class CrashReportPage(ErrorReportUIBase, ErrorReportUI):
  File "/home/3y9/Mantid/mantid/qt/python/mantidqt/dialogs/errorreports/report.py", line 34, in CrashReportPage
    interface_manager = InterfaceManager()
Exception: unknown
```

### Steps to reproduce the behavior
Add `instrumentDefinition.directory=./instrument` to the Mantid.user.properties file

### Platforms affected
At least OSX and Ubuntu 20.04

martyngigg, 2021-05-05:
It's quite likely that this is related to #31354. That issue has been assigned so we can hold off looking at this.

martyngigg, 2021-05-10:
Fixed by #31376

PR #31376 (2021-05-10): Fix ConfigService crash when using an invalid instrumentDefinition.directory
**Description of work.**
This PR fixes a crash caused by an invalid instrument directory when the ConfigService singleton is created. If this directory is invalid, the `%appdata%` instrument directory is the only remaining directory we can use, and so the `instDir` iterator should not be incremented in this case.

It is difficult to add a test to cover this bug as the bug would only occur when the ConfigService singleton is constructed (i.e. this happens only once). The offending line would need to be in a Mantid.user [... cut]
closed 2021-05-10 by martyngigg