mantidproject/mantid #30642: MacOS Smoke Tests: Release 6.0

DanielMurphy22, 2021-02-09:
Before testing:
- Check this master issue relates to the OS you will test on.
- Find an unassigned testing issue below and assign yourself.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment on the testing issue with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.

If you have any questions please contact the creator of this issue.

## 1. MacOS Clean/Dirty Install Smoke Tests :soap: :hankey: #30636

## 2. MacOS Fitting Smoke Tests :muscle: :runner: #30637

## 3. MacOS Interface Smoke Tests :facepalm: #30638

## 4. MacOS Plotting Smoke Tests :chart_with_upwards_trend: :bar_chart: #30639

## 5. MacOS Python Smoke Tests :snake: #30640

## 6. MacOS Workspace/Algorithm Smoke Tests  :man_astronaut: :woman_astronaut: :space_invader: #30641

martyngigg, 2021-02-09:
All of these are now closed.

closed 2021-02-09 by martyngigg

gvardany, 2021-02-10:
Just tried on the newer version of the OS Big Sur and there it doesn't work.
This was already reported by some users.
The applications just wouldn't start and freeze to a non-responding state without a trace.
Any chance we could try to reproduce/debug?
It might be a trivial issue again with the command line arguments, albeit non-trivial to find.
Big Sur was out in November 2020, so many users have already migrated, would be nice to make the release supported on that.

martyngigg, 2021-02-10:
Did the beta version work on it or the previous version?

martyngigg, 2021-02-10:
Does starting from the command line show anything? - https://developer.mantidproject.org/UserSupport.html#macos