mantidproject/mantid #30651: Ubuntu Fitting Smoke Tests

DanielMurphy22, 2021-02-09:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :muscle: :runner:

* Fitting to data [directions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/01_fitting_models_to_data.html#fitting-models-to-data)
- [ ] Simple fit 
- [ ] Activating the fitting tools 
- [ ] Selecting the fitting range 
- [ ] Other Fit Function Settings properties 
- [ ] Setting up a fit model 
- [ ] Adjusting fit function parameter 
- [ ] Saving a model 
- [ ] Tying and constraining fit parameters 
- [ ] Execute your fit 
- [ ] Inspecting a fit result 
* Additional options [directions](https://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/fitting_data/03_fit_model_choices.html)
- [ ] Non peak model + background 
- [ ] User defined function

RichardWaiteSTFC, 2021-02-09:
The ExcludeRange option is missing from the FitPropertyBroswer on ubuntu
[image]

RichardWaiteSTFC, 2021-02-09:
There is a problem with copying constraints to a string. I had a model with two Gaussians and a background. I fixed one bg param and constrained the other, then added some constraints to the peak function (peak centres were opposite sign). Copying the setup to a string I got
```
composite=CompositeFunction,NumDeriv=true;name=Gaussian,Height=0.983135,PeakCentre=-2.99456,Sigma=0.402576;name=Gaussian,Height=0.979642,PeakCentre=2.99456,Sigma=0.408717;name=LinearBackground,A0=0.0499696,A1=-0.000829793,constraints=(A0<0.05),ties=(A1=-0.000829793);ties=(f1.PeakCentre=-f0.PeakCentre)
```
When saving this as a setup, or loading this string back into the fit browser the constraint on the Gaussian centres was not applied - perhaps because they appear in a different function where the ties to the background are written.  For example when I fix sigma of one peak I get the following string
```
composite=CompositeFunction,NumDeriv=true;name=Gaussian,Height=0.929024,PeakCentre=-2.9947,Sigma=0.437051,ties=(Sigma=0.437051);name=Gaussian,Height=0.988602,PeakCentre=2.9947,Sigma=0.409895;name=LinearBackground,A0=0.0323165,A1=0,constraints=(A0<0.05),ties=(A1=0);
```
I see the same thing on windows master.

martyngigg, 2021-02-09:
> The ExcludeRange option is missing from the FitPropertyBroswer on ubuntu
> [image]
> 
> Setup > Manage Setup > Clear Setups is also missing from ubuntu

This is odd. Has it appeared before?

martyngigg, 2021-02-09:
> There is a problem with copying constraints to a string. I had a model with two Gaussians and a background. I fixed one bg param and constrained the other, then added some constraints to the peak function (peak centres were opposite sign). Copying the setup to a string I got
> 
> ```
> composite=CompositeFunction,NumDeriv=true;name=Gaussian,Height=0.983135,PeakCentre=-2.99456,Sigma=0.402576;name=Gaussian,Height=0.979642,PeakCentre=2.99456,Sigma=0.408717;name=LinearBackground,A0=0.0499696,A1=-0.000829793,constraints=(A0<0.05),ties=(A1=-0.000829793);ties=(f1.PeakCentre=-f0.PeakCentre)
> ```
> 
> When saving this as a setup, or loading this string back into the fit browser the constraint on the Gaussian centres was not applied - perhaps because they appear in a different function where the ties to the background are written (although when I move it I get an error saying f0 doesn't have a parameter PeakCentre). For example when I fix sigma of one peak I get the following string (which does fix sigma when reloaded).
> 
> ```
> composite=CompositeFunction,NumDeriv=true;name=Gaussian,Height=0.929024,PeakCentre=-2.9947,Sigma=0.437051,ties=(Sigma=0.437051);name=Gaussian,Height=0.988602,PeakCentre=2.9947,Sigma=0.409895;name=LinearBackground,A0=0.0323165,A1=0,constraints=(A0<0.05),ties=(A1=0);
> ```
> 
> I see the same thing on windows master.

Sounds like this a long standing bug that won't hold up the release.

RichardWaiteSTFC, 2021-02-09:
The issue with constraints in the function string existed in 4.2 so definitely not something to hold up the release if it hasn't been reported as a bug.

DanielMurphy22, 2021-02-09:
> > The ExcludeRange option is missing from the FitPropertyBroswer on ubuntu
> > [image]
> > Setup > Manage Setup > Clear Setups is also missing from ubuntu
> 
> This is odd. Has it appeared before?

Checked Ubuntu Workbench 4.2 and ExcludeRange isn't displayed there either, so I think this is a Windows vs Ubuntu difference but not new to the current release candidate.

martyngigg, 2021-02-09:
I can see it in 5.1.1 that I have on my Ubuntu desktop. Just trying 6.0 now.

martyngigg, 2021-02-09:
I also see it in my 6.0 version too. Is it possible the original Ubuntu one is actually a 4.2?

RichardWaiteSTFC, 2021-02-09:
Yes just checked and it was v4.2...my installation of v6.0 failed and didn't remove the previous one (which I'd forgot I'd installed). Sorry for the confusion, I'll try again tomorrow morning! HAs anyone else reported problems with the .deb package not installing in ubuntu?

martyngigg, 2021-02-10:
No problem. I'd rather that there weren't an issue :)

I was able to install the candidate myself and I think @Pasarus was able to?

Pasarus, 2021-02-10:
Mine was fine
I didn't over/dirty install such an older version though.

DanielMurphy22, 2021-02-10:
I just did a dirty install of 6 over 4.2 using the manual install instructions at the bottom [here](https://download.mantidproject.org/ubuntu.html) and the install worked fine.

RichardWaiteSTFC, 2021-02-10:
Sorry everyone, I didn't realise there were specific instructions to follow for installing on Ubuntu - I was just opening the .deb with software manager...since @DanielMurphy22 has successfully installed 6.0 over 4.2 I think there's no problem.

DanielMurphy22, 2021-02-10:
We should probably include the install instructions on the Smoke test issues next time. Easy to forget Linux is different.

closed 2021-02-10 by DanielMurphy22