mantidproject/mantid #30653: Ubuntu Plotting Smoke Tests

DanielMurphy22, 2021-02-09:
Before testing:
- Check this testing issue relates to the OS you will test on.
- If unassigned, please assign yourself as for a normal Github issue.
- Please run these tests on the release package of Mantid; **not a locally built version**.

Afterwards:
- Comment below with any issues you came across.
- If no issues were found, or they are now all resolved, please close the testing issue.
- Check the master issue for this OS for other unassigned smoke tests.

If you have any questions please contact the creator of this issue.

 :chart_with_upwards_trend: :bar_chart:

* 1D plotting:[instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/03_displaying_1D_data.html#displaying-1d-data)
 - [ ] Simple plot 
 - [ ] Another way to plot 
 - [ ] Adding curves to existing plots 
 - [ ] Also, test out [waterfall](https://docs.mantidproject.org/nightly/plotting/WaterfallPlotsHelp.html#waterfall-plots) and [tiled]( https://docs.mantidproject.org/nightly/plotting/1DPlotsHelp.html#tiled-plots)
 - [ ] Check Toolbar buttons

* 2D plotting: [instructions](http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/loading_and_displaying_data/04_displaying_2D_data.html#displaying-2d-data)
 - [ ] Plot all spectra 
 - [ ] Change colour map 
 - [ ] [Contour plot](https://docs.mantidproject.org/nightly/plotting/ColorfillPlotsHelp.html#contour-plots) (under 3D menu) 
 - [ ] Check Toolbar buttons

* [3D plotting](https://docs.mantidproject.org/nightly/plotting/3DPlotsHelp.html):
 - Load some data eg `LOQ7041` from the ISIS sample data
 - [ ] 3D surface
 - [ ] 3D wire frame  
 - [ ] Check Toolbar buttons

 ## Sliceviewer
 - [ ] Overly long instructions (don't spend hours!) and data [here](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html). In particular try editing the data in a workspace while it is open in Sliceviewer! ([step 10](https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#alter-the-underlying-workspace))

martyngigg, 2021-02-10:
Are we happy with the state of this for the release?

MathieuTillet, 2021-02-10:
In the slice viewer, if I show 2D data and unzoom by right clicking a box, I have a lot of errors appearing. The slice viewer doesn't crash and behave well, but that's not optimal.

MathieuTillet, 2021-02-10:
Also on 2D data, right-clicking the plot completely mess with the color scale.

Apart from these two interface bugs on the slice viewer, I found nothing of concern.

RichardWaiteSTFC, 2021-02-10:
No error on v5.1.1 in ubuntu 18.04 on right-clicking to unzoom data in sliceviewer - but the colorbar does rescale. I think it's because the autoscaling of the colorscale when the data is first loaded isn't working correctly - it looks a bit saturated already
[image]

DanielMurphy22, 2021-02-10:
I think the error message for both of these issues is as below:
```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py", line 388, in process
    proxy(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/matplotlib/cbook/__init__.py", line 228, in __call__
    return mtd(*args, **kwargs)
  File "/opt/Mantid/lib/mantidqt/widgets/sliceviewer/cursor.py", line 65, in on_mouse_move
    self.on_cursor_at(xdata, ydata)
  File "/opt/Mantid/lib/mantidqt/widgets/sliceviewer/imageinfowidget.py", line 66, in _on_cursor_at_axesimage
    self._widget.cursorAt(xdata, ydata, arr[i, j])
RuntimeError: indexOfValue - value out of range
```
I assume it's only an Ubuntu problem as the Matplotlib version is behind Windows+MacOS.

I think this is a little separate to the error, but just right-clicking on the image in sliceviewer is trying to rescale the axis (equivalent to clicking the home toolbar button).

MathieuTillet, 2021-02-10:
This is the first error message, yes.
For the second one, just right clicking tries to rescale the axis, but rescales it to some value I can't find the source of.

DanielMurphy22, 2021-02-10:
> For the second one, just right clicking tries to rescale the axis, but rescales it to some value I can't find the source of.

I'm not suggesting it's correct, but does clicking the home toolbar button go to the same value as right-clicking?

MathieuTillet, 2021-02-10:
Ok, sorry. Yes, the home button has the same behaviour, but the values seems to depend on where the zoom took place.
More precisely, if you zoom somewhere then right click or click the home button, the axis is rescaled to a value, and then if you click again the axis goes to its correct (or at least final) values (and the data shown is updated accordingly).

It looks like the color scale is updated based on the previous values, not the current ones.

martyngigg, 2021-02-11:
Can we create issues for these so we don't lose the information on what we need to fix going forward?

closed 2021-02-11 by martyngigg

DanielMurphy22, 2021-02-11:
Raised https://github.com/mantidproject/mantid/issues/30677 and https://github.com/mantidproject/mantid/issues/30678