mantidproject/mantid #30681: MacOS BigSur mantid will not open

DanielMurphy22, 2021-02-11:
**Original reporter:** [Mantid Help Forum](https://forum.mantidproject.org/t/mantid-crash-on-startup-in-big-sur-macos-11-0-1/707/14) and @gvardany :https://github.com/mantidproject/mantid/issues/30642#issuecomment-776520506

After upgrading to MacOS Big Sur workbench packages for 5.1.1 and the release candidate for 6.0 both do not open.

Release 5.0 opens and locally built packages have worked too. This seems to be a Qt version issue: https://bugreports.qt.io/browse/QTBUG-87014

@martyngigg

gui-co, 2021-02-11:
Here is a workaround mentioned in the Qt bug report that worked for @gvardany:
```bash
export QT_MAC_WANTS_LAYER=1
```
As far as I understood, the problem should disappear if the application is build with Xcode >= 10

DanielMurphy22, 2021-02-12:
As mentioned by Ian on the forum, this workaround seems to apply to macs with an M1 chip.

martyngigg, 2021-04-06:
The referenced [Qt bug report](https://bugreports.qt.io/browse/QTBUG-87014) suggests the Qt fixes for this issue have been committed to both the 5.12 & 5.15 branches but unfortunately these fixes are not in an officially released patch version.

The diff for the [fix](https://codereview.qt-project.org/gitweb?p=qt/qtbase.git;a=commitdiff;h=c5d904639dbd690a36306e2b455610029704d821) suggests that building with the correct 10.14 SDK would help - we seem to already be building with Xcode 10.1. The current SDK is 10.13 but looks like it has always been that.

martyngigg, 2021-04-06:
As moving Qt version is hard anyway due to homebrew `HEAD` moving things around and it having a very new version of `sip` that requires bigger changes for us to cope with I wondered whether we should do our own detection of the macOS runtime in [mantidworkbench.cpp](https://github.com/mantidproject/mantid/blob/8b5ed231e3c06dd6b5097c040cff64f7570a1c1c/qt/applications/workbench/mantidworkbench.cpp) and set the `QT_MAC_WANTS_LAYER` manually?

The other option might be to see if we can set [CMAKE_OSX_DEPLOYMENT_TARGET](https://cmake.org/cmake/help/latest/variable/CMAKE_OSX_DEPLOYMENT_TARGET.html) to 10.14.

PR #31159 (2021-04-21): Workaround a Qt bug on macOS Big Sur
**Description of work.**

The fix for Qt has yet to released so we need
to force layer-backed windows on Big Sur
and above

**To test:**

* Build a package on the CI servers
* Test the built package starts on Big Sur

Fixes #30681

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of an acceptable quality?
- Does the code conform to the [coding standards](http://developer.mantidproject.org/Standards/)?
- Are the unit tests smal [... cut]
closed 2021-04-21 by DavidFair