mantidproject/mantid #30738: Instrument View: Display additional informational cursor information for assembly selections

martyngigg, 2021-02-22:
As a user of the instrument view I want to know the counts over a whole tube of an instrument but the selection pane In pick mode tab on the instrument view shows details of the pixel the cursor is positioned over only.

The information displayed always uses the single pixel over which the cursor hovers, regardless of the pick mode activated, e.g. tube selection. It has been highlighed by users that this can be a little misleading as the plotting pane shows the sum in the case of tube selection for example. The following image shows an example:

[image]

This task is to add information to selection pane giving addtional details on the sum of the counts the cases where multiple pixels are selected in the plotting pane.

thomashampson, 2021-03-03:
I propose that we rename "Counts" to "Pixel counts" and add an extra line with "Tube counts" when the user is in tube selection mode.

martyngigg, 2021-03-04:
This sounds like a sensible suggestion to me.

PR #30900 (2021-03-22): 30738 tube counts instrument view
**Description of work.**

Added additional information to the selection pane in the pick tab of the instrument view. When using the tube selection tool, "Tube counts" are now shown. This is a sum of all of the pixel counts within the parent of the pixel that is currently hovered over.

**To test:**

- Load a workspace and open the instrument view.
- Go to the pick tab.
- Using the pixel selection tool, hover over a pixel. The pixel count will be shown in the selection pane.
- Using the tube selection tool, hover over a pixel. The tube count is also shown.

Fix [... cut]
closed 2021-03-22 by gvardany