mantidproject/mantid #30779: Segfault in PaalmanPingsMonteCarloAbsorption doc test example

martyngigg, 2021-02-26:
### Expected behavior

All tests pass using the default number of cores.

### Actual behavior

In #30773 the number of cores was reduced to 1 to avoid what looks like a memory corruption issue.

### Steps to reproduce the behavior

On my mac with a release or debug build I ran the following script:

```python
from mantid.simpleapi import *

def run():
    sample_ws = CreateSampleWorkspace(Function="Quasielastic",
                                      XUnit="Wavelength",
                                      XMin=-0.5,
                                      XMax=0.5,
                                      BinWidth=0.01)
    # Efixed is generally defined as part of the IDF for real data.
    # Fake it here
    inst_name = sample_ws.getInstrument().getName()
    SetInstrumentParameter(sample_ws, ComponentName=inst_name,
        ParameterName='Efixed', ParameterType='Number', Value='4.1')

    # define example geometries for the Sample and Container
    SetSample(sample_ws, Geometry={'Shape': 'Cylinder', 'Height': 4.0, 'Radius': 2.0, 'Center': [0.,0.,0.]},
                Material={'ChemicalFormula': 'Ni', 'MassDensity': 7.0},
                ContainerGeometry={'Shape': 'HollowCylinder', 'Height': 4.0, 'InnerRadius': 2.0,
                'OuterRadius': 3.5})

    corrections = PaalmanPingsMonteCarloAbsorption(
            InputWorkspace=sample_ws,
            Shape='Preset',
            BeamHeight=2.0,
            BeamWidth=2.0,
            CorrectionsWorkspace='annulus_corr'
        )

    # alternatively, run the corrections with the sample material overridden but preset shape used
    corrections = PaalmanPingsMonteCarloAbsorption(
            InputWorkspace=sample_ws,
            Shape='Preset',
            BeamHeight=2.0,
            BeamWidth=2.0,
            SampleChemicalFormula='H2-O',
            SampleDensity=1.0,
            CorrectionsWorkspace='annulus_corr'
        )

    ass_ws = corrections[0]
    assc_ws = corrections[1]
    acsc_ws = corrections[2]
    acc_ws = corrections[3]

    print("Y-Unit Label of " + str(ass_ws.getName()) + ": " + str(ass_ws.YUnitLabel()))
    print("Y-Unit Label of " + str(assc_ws.getName()) + ": " + str(assc_ws.YUnitLabel()))
    print("Y-Unit Label of " + str(acsc_ws.getName()) + ": " + str(acsc_ws.YUnitLabel()))
    print("Y-Unit Label of " + str(acc_ws.getName()) + ": " + str(acc_ws.YUnitLabel()))

run()
```

### Platforms affected

macOS segfaults but I suspect the bug is just more hidden on other systems. The santizers might be able to help.

I also have a stacktrace from the crashed run:
[... 1060 lines cut ...]
Activity Tracing                   256K        1 
Dispatch continuations            32.0M        1 
Kernel Alloc Once                    8K        1 
MALLOC                           223.1M       54 
MALLOC guard page                   24K        5 
MALLOC_MEDIUM (reserved)           1.0G        9         reserved VM address space (unallocated)
STACK GUARD                         68K       17 
Stack                            136.6M       17 
VM_ALLOCATE                      144.8M      192 
VM_ALLOCATE (reserved)           384.0M        2         reserved VM address space (unallocated)
__DATA                            30.7M      562 
__DATA_CONST                      7492K      118 
__FONT_DATA                          4K        1 
__LINKEDIT                         1.0G      283 
__OBJC_RO                         32.3M        1 
__OBJC_RW                         1908K        2 
__TEXT                           536.6M      489 
__UNICODE                          564K        1 
mapped file                       27.3M        2 
shared memory                       44K        5 
===========                     =======  ======= 
TOTAL                              3.6G     1763 
TOTAL, minus reserved VM space     2.2G     1763 
```
  </details>

DannyHindson, 2021-03-03:
This seg fault is also occasionally happening in the Jenkins jobs for pull requests during one of the unit tests for the PaalmanPingsMonteCarloAbsorption algorithm:
https://builds.mantidproject.org/job/pull_requests-ubuntu/42670/
https://builds.mantidproject.org/job/pull_requests-ubuntu/42667/

These were the only two jobs on 1st or 2nd March that failed so not very often. It was also showing in the doc test on Ubuntu but the change to restrict running to a single thread seems to have cleared that up.

I don't see it happening on the Windows PR jobs for some reason. Will see if I can reproduce on Ubuntu or whether the address sanitizer shows anything on Windows

DannyHindson, 2021-03-09:
This seg fault seems to be caused by a call to new in the multithreaded part of the MonteCarloAbsorption algorithm. If the sample or environment contain some CSG shapes that don't have a material defined by the user then the function CSGObject::material creates a dummy material. The materials on the sample\environment shapes are shared between threads so perhaps more than one thread created a new material occasionally if the threads both entered CSGObject::material at the same time.

The doc test that was added for PaalmanPingsMonteCarloAbsorption (which calls MonteCarloAbsorption) doesn't define a material for the container so that introduced an example of this situation. Two unit tests in PaalmanPingsMonteCarloAbsorptionTest (test_preset_without_overriding_material and test_preset_with_overriding_material) also ran MonteCarloAbsorption without a material for the container.

This was quite a pain to track down because the seg fault seemed to happen some time after the call(s) to CSGObject::material that created the dummy material. The call to MonteCarloAbsorption completed normally most times and on the next call to an algorithm (any algorithm), a call to new in a completely unrelated part of the code would seg fault. Seems like the heap was corrupted.

closed 2021-03-23 by gvardany