mantidproject/mantid #30792: Plan move to conda dependency management

martyngigg, 2021-03-01:
It was decided by the TSC that the [dependency management](https://github.com/mantidproject/documents/pull/78) would transition over to a conda-first approach for managing our dependencies rather than the 3 systems we currently employ for this over the 3 different operating systems.

This task is to plan how the rollout of this switch.

marshallmcdonnell, 2021-03-11:
Update:
For reviewing PR https://github.com/mantidproject/mantid/pull/30859, I plan to go ahead and setup an `environment.yml` file based off of the conda recipe [`meta.yml`](https://github.com/mantidproject/conda-recipes/blob/master/framework/meta.yaml)

This would be related to _"how developers will interact with it"_ above.

Pasarus, 2021-03-23:
| Package                        | Windows                  | MacOS     | Ubuntu                     | CentOS7/RHEL7 | Chosen Version           | Conda supported version |
|--------------------------------|--------------------------|-----------|----------------------------|---------------|--------------------------|-------------------------|
| Python                         | 3.8                      | 3.8       | 3.6                        | 3.6.8         | 3.8.8                    | 3.9.2                   |
| g++                            | NA                       | NA        | 7.5.0                      | 4.8.5         | 9.3.0                    | 9.3.0                   |
| LLVM                           | NA                       | ?         | NA                         | NA            | ?                        | 11.1.0                  |
| clang-format                   | 6.0                      | 6.0       | 6.0                        | 6.0           | Not needed w/ pre-commit | 5.0.1                   |
| cmake                          | ?                        | ?         | 3.19.5                     | 3.17.5        | 3.19.7                   | 3.19.7                  |
| dvipng / MikTex                | NA / 2.9.6753            | 1.17 / NA | 1.15 / NA                  | 1.14 / NA     | 1.17 / 2.9.6753          | NA / 2.9.6753           |
| doxygen                        | ?                        | 1.9.1     | 1.18.13                    | 1.8.9.1       | 1.9.1                    | 1.9.1                   |
| tbb-dev                        | tbb2019_20191006oss_win. | 2020_U3   | 2017~U7-8                  | 4.1           | 2021.1.1                 | 2021.1.1                |
| perftools                      | ?                        | 2.9.1     | 2.5-2.2ubuntu3             | 2.6.1         | 2.9.1                    | 2.7 (No Win)            |
| boost-all-dev                  | 1.63.0                   | 1.75.0    | 1.65.1.0ubuntu1            | 1.53.0-28.el7 | 1.75.0                   | 1.75.0                  |
| poco-dev                       | 1.6.1                    | 1.10.1    | 1.8.0.1-1ubuntu4           | 1.6.1-3.el7   | 1.10.1                   | 1.10.1                  |
| nexus-dev                      | master                   | 4.4.3     | 4.3.2                      | 4.3.3-1.el7   | 4.4.3                    | 4.4.3 (No Win)          |
| libhdf5-dev                    | 1.10.5                   | ?         | 1.10.0-patch1              | 1.8.12-11.el7 | 1.12.0                   | 1.12.0                  |
| libhdf4-dev                    | ?                        | ?         | 4.2.13-2                   | 4.2.13-1.el7  | 4.2.13                   | 4.2.13                  |
| libjemalloc-dev                | ?                        | 5.2.1     | 3.6.0-11                   | 3.6.0-1.el7   | 5.2.1                    | 5.2.1                   |
| libgsl-dev                     | 2.5                      | 2.6       | 2.4+dfsg-6                 | 1.15-13.el7   | 2.6                      | 2.6                     |
| libjsoncpp-dev                 | 1.8.1                    | 1.9.4.1   | 1.7.4-3                    | 0.10.5-2.el7  | 1.9.4                    | 1.9.4                   |
| librdkafka-dev                 | 1.1.0                    | 1.6.1     | 0.11.3-1build1             | 0.11.5-1.el7  | 1.6.1                    | 1.6.0 (No Win)          |
| qt5                            | 5.8                      | 5.15.2    | 5.9.5                      | 5.9.7.el7     | 5.15.2                   | 5.12.9                  |
| qt5-tools                      | "                        | "         | 5.9.5                      | 5.9.7.el7     | ?                        | ?                       |
| Qtwebkit                       | "                        | "         | 5.212.0~alpha2-7ubuntu1    | 5.9.1-2.el7   | 5.212                    | 5.212                   |
| libqscintilla2-qt5-dev         | "                        | "         | 2.10.2+dfsg-4              | 2.9.3-1.el7   | 2.11.2                   | 2.11.2                  |
| libpython3-dev                 | ?                        | ?         | 3.6.7-1~18.04              | 3.6.8-18.el7  | ?                        | ?                       |
| ninja-build                    | NA                       | ?         | 1.8.2-1                    | 1.7.2-4.el7   | 1.10.2                   | 1.10.2                  |
| python3-setuptools             | 41.2.0                   | 48.0.0    | 39.0.1-2                   | 39.2.0-10.el7 | 48.0.0                   | 50.0.0                  |
| python3-sip-dev                | 4.19.13                  | ?         | 4.19.7+dfsg-1ubuntu0.1     | 4.19.22-1.el7 | 4.19.18                  | 4.19.18                 |
| python3-pyqt5                  | 5.11.3                   | 5.15.2    | 5.10.1+dfsg-1ubuntu2       | 5.12.1-3.el7  | 5.12.3                   | 5.12.3                  |
| pyqt5-dev                      | 5.11.3                   | 5.15.2    | 5.10.1+dfsg-1ubuntu2       | 5.12.1-3.el7  | "                        | "                       |
| pyqt5-dev-tools                | 5.11.3                   | 5.15.2    | 5.10.1+dfsg-1ubuntu2       | 5.12.1-3.el7  | "                        | "                       |
[... cut ...]
| python3-toml                   | 0.10.0                   | 0.10.0    | 0.9.3-1                    | 0.9.4-3.el7   | 0.10.2                   | 0.10.2                  |
| python3-yaml                   | 5.2                      | 5.3.1     | 3.12-1build2               | 3.13-1.el7    | 5.4.1                    | 5.4.1                   |
| pre-commit                     | ?                        | ?         | ?                          | ?             | 2.11.1                   | 2.11.1                  |

StephenSmith25, 2021-05-21:
Meeting 21/05/2021 - (Martyn, Stephen, Sam and Rob)

- Discussed the progress with the Conda environments. We currently have a working Conda environment for all three OSes that is capable of building a Mantid package. There are two caveats:
    1.  On Mac OSX you have to specify a few CMake arguments, namely turning dynamic linking to python off and specifying the prefix path.
    2.  On windows we are missing the Fortran routines, which are exposed to python using  f2py. With the availability of Fortran compilers in Conda, we discussed eventually supporting this feature in all three builds by pulling the Fortran code in as a CMake project.
- Sam is currently fixing issues relating to JSONCpp on Linux. Once this is merged, the Mac and Windows environments will be submitted. 
- We discussed the developer interaction. Currently, a new developer will need to download Miniconda, add Conda-forge to their channels and then activate the relevant Conda environment. 
- We discussed the pinning of Conda packages. One of the goals of the Conda package will be to allow users to integrate Mantid into their workflow, which may feature other python packages. To support this, we need to consider how we pin/specify our dependencies in a way that ensures Mantid can be used alongside other popular packages, e.g. Pandas.

Pasarus, 2021-05-28:
I have converted this to an Epic using ZenHub and the following tasks will have issues made and assigned for it.
- Linux Developer Environment
- OSX Developer Environment
- Windows Developer Environment
- Developer Documentation (Getting Started Guide)
- Make the Jenkins pipeline for conda using a developer environment
- Conda training session planning, and development
- Debug build type in windows that is functional with release ABIs (i.e. a release build with no optimizations turned on, with debug info)
- Define what the packaging strategy is in a Markdown document so other facilities can discuss it
- Conda Build implementation of Mantid so anyone can get Mantid from the Anaconda Cloud

Pasarus, 2022-02-03:
This epic is now retired and some new epics have been spawned to create the deliverables discussed here.

closed 2022-02-03 by Pasarus