mantidproject/mantid #30796: Investigate updating focussed data with instrument parameters post EditInstrumentGeometry

RichardWaiteSTFC, 2021-03-01:
The Engin-X calibration involves focussing a bank and then updating the instrument geometry so that there is a single detector at some representative difc  so that when the data are transformed back into TOF then  difa = tzero = 0 (though these are subsequently calibrated again). 
However, overwriting the instrument geomtery means the workspace no longer knows about the BackToBackExponential parameters - it is possible to use `LoadParameterFile` is the instrument neame is set to the name of the components in the Parameters.xml file (e.g. NorthBank). But this does not survive whent he workspace is saved (i.e. this will need to be done each time) and mantid complains NorthBank isn't an instrument.

RichardWaiteSTFC, 2021-03-30:
DiffractionFocussing seems to produce a table with avg L2 and tth - still looking into this to confirm 
[image]
We can use ConvertUnits to transform back to TOF (by default DIFA=TZERO=0) and then calibrate using PDCalibration, ApplyDiffCal and not overwrite geometry

```
Load(Filename=r'\\isis.cclrc.ac.uk\inst$\ndxenginx\instrument\data\cycle_12_4\ENGINX00193749.nxs', OutputWorkspace='193749')
ConvertUnits(InputWorkspace='193749', OutputWorkspace='193749', Target='dSpacing')
DiffractionFocussing(InputWorkspace='193749', OutputWorkspace='193749_foc', GroupingFileName='C:/Users/xhg73778/OneDrive - Science and Technology Facilities Council/EnginX_NorthBank.cal')
ConvertUnits(InputWorkspace='193749_foc', OutputWorkspace='193749_foc_TOF', Target='TOF')
PDCalibration(InputWorkspace='193749_foc_TOF', TofBinning='15500,-0.0003,52000', PeakFunction='BackToBackExponential', PeakPositions='2.7057,1.91322,1.6316,1.56214,1.35285,1.24146,1.21003,1.1046,1.04143,0.95661,0.914694,0.901901,0.855618', PeakWindow=0.040000000000000001, MinimumPeakHeight=0.5, CalibrationParameters='DIFC+TZERO+DIFA', UseChiSq=True, OutputCalibrationTable='cal_North', DiagnosticWorkspaces='diag_North')
ApplyDiffCal(InstrumentWorkspace='193749_foc_TOF', CalibrationWorkspace='cal_North')
```
[image]

DannyHindson, 2021-03-30:
Closing - will cover in separate issue on updating Engg calibration

closed 2021-03-30 by DannyHindson