mantidproject/mantid #30813: Export fails on drill

gvardany, 2021-03-04:
Exporting in drill fails under certain circumstances.
This could be related to the fact, that the ws is a group, so you can't get its history directly.
Also there must be a check if history exists, and if it is not empty.
In any case failure of exports shouldn't result into crying mantid.

### Steps to reproduce the behavior
Start drill. Set datasearch directory to current cycle internalUse of D11.
Put 000560 as sample, check OutputPanels in global settings.
Click process > 💣

gui-co, 2021-03-04:
This is due to the fact that there is apparently no history in panel workspaces. The history is used to check that the output type is compatible with the selected export algorithm.

gvardany, 2021-03-04:
I see, the reason for this is that panels are not output workspaces of the algorithm, but they are generated when there is a checkbox. I think this can be fixed if we declare the output properties at runtime. However, in the save side, cases when there is no history, or no last item in it should be properly guarded.

PR #30841 (2021-03-19): Bug in DrILL export
**Description of work.**

During the automatic export in DrILL, criteria are applied on workspaces to verify that a specific export algorithm can be applied on the data. This is using the processing parameters found in the workspace history. This was causing a crash with workspaces with empty history (for example, panel workspaces in D11 data processing).

To resolve this problem, the algorithm history is now in a try-catch block and the criteria are validated once per sample on the output workspace.

**To test:**

Try to reproduce the issue #30813 

Fixes #30813 

---

### [... cut]
closed 2021-03-19 by martyngigg