mantidproject/mantid #30827: Implement a widget for workspace calculator

darominski, 2021-03-05:
ILL users requested a calculator for quick calculations of workspaces that include 4 basic operations:summation, subtraction, multiplication, and division. Also, scaling of both LHS and RHS workspaces by a float number was requested.

### Platforms affected
Workbench on all platforms

SpencerHowells, 2021-03-05:
@darominski What is wrong with the algorithms Plus, Minus, Multiply, Divide & Scale?

gvardany, 2021-03-05:
Spencer, this is just a convenience widget that does 3 operations with one click: scale both operands and run a binary operation. Even if this is a one-liner in python, the widget is for those users allergic to code.

SpencerHowells, 2021-03-08:
@SpencerHowells Gagik, you don't need to know Python I regularly use the alg dialog window. I also have the interface Inelastic>corrections>Subtraction which also does ScaleX to do a shift. Also gives instant plots of result!

darominski, 2021-03-09:
@SpencerHowells I agree one does not need to know Python to use Mantid, however this feature is intended to improve the experience of users currently reluctant to directly interact with algorithms through dialogs or write scripts to perform these quick operations on workspaces. This is the way we intend to use to ease their transition into using Mantid.

SpencerHowells, 2021-03-09:
@darominski I thought that might be the case. We also have trouble with users reluctant to use Mantid properly. Wouldn't algebraic manipulation be easier eg w = 2*w1 - w2/3 ?

closed 2021-06-22 by martyngigg