mantidproject/mantid #30861: User feedback to D7YIGPositionCalibration

darominski, 2021-03-10:
Following the user feedback certain modifications to the algorithm and its documentation are needed:

1. Fix unwanted renaming of the workspace provided through `InputWorkspace` property.
2. Add loading feature for the InputWorkspace so that it doesn't need to be pre-loaded each time.
3. Include Chi^2 value in the fit output information.
4. Update description for `MaskedBinRange` property to showcase the workflow of masking desired 2theta ranges.

gvardany, 2021-03-11:
I'd add here to check why the initial bank offsets were not taken in the account.

darominski, 2021-03-11:
> I'd add here to check why the initial bank offsets were not taken in the account.

I didn't add it because I couldn't reproduce the issue neither on VISA nor on my local build, with a variation of different input parameters. I believe the issue was caused by using too small of an offset, 3 degrees was not sufficient, as the bank2 and bank3 seem to be out of expected position by more than 4 degrees. The feature actually works as intended, but the workflow needs to be explained better to the user.

gvardany, 2021-03-11:
But we didn't see the 3 degree in the noop mode (i.e. no fitting). Are we not supposed to directly see the shift there?

darominski, 2021-03-11:
> But we didn't see the 3 degree in the noop mode (i.e. no fitting). Are we not supposed to directly see the shift there?

Not necessarily, because in this mode the algorithm provides us the initial guesses for the fitting. This includes looking for a local maximum in a window (say, 1.5 degrees, defined by user) around the expected position to obtain the height and mean for the Gaussian to be fitted. 3 degrees was not sufficient to cause a shift, that part I could reproduce locally, while 4 degrees and above would cause a visible change in which peaks are (going to be) fitted.

martyngigg, 2021-04-29:
Fixed by #30910

closed 2021-04-29 by martyngigg