mantidproject/mantid #30863: User feedback to PolDiffILLReduction technique

darominski, 2021-03-10:
Following the user feedback certain modifications to the `PolDiffILLReduction` and `D7AbsoluteCrossSections` algorithms and to the relevant documentation of the technique are needed:

- [ ] Instrument calibration coming from YIG calibration is not specified in the examples of the `PolDiffILLReduction` nor the `PolarizedILLDiffraction` technique documentation
- [ ] Sample height, beam height and beam width are not parameters commonly known: their value should be fixed so it does not impact attenuation calculation, if possible.
- [ ] The density choice between number density and mass density is not clear enough.
- [ ] The unit for Y-axis of polarization correction is incorrect
- [ ] Flipping ratio plots should be made available for instrument scientist, along with polarization correction value
- [ ] The algorithm crashes when processing vanadium in `PolDiffILLReduction` with the property `OutputTreatment` set to 'Average'
- [ ] Add option to turn off the normalisation for diagnostic purposes
- [ ] `D7AbsoluteCrossSections` does not provide expected results when either 'Sum' or 'Average' are specified for the `OutputTreatment` property
- [ ] There is an issue in calculating cross-sections of magnetic samples: the magnetic component cross-section looks inverted, which leads to issues in incoherent scattering cross-section.

gvardany, 2021-03-11:
Please add an item to double check the error propagation throughout.

darominski, 2021-05-04:
Closed with the linked PR.

closed 2021-05-04 by darominski