mantidproject/mantid #30883: PEARL output file requests - Nice to have

DanielMurphy22, 2021-03-12:
**Original reporter:** Nick Funnell

- 1) With `focus_mode=‘All’`, all file names are appended with a “-0”. The -0 signifies merged banks 1-9, so perhaps the -0 could be dropped when it’s a single bank, e.g. `XXXXXX_tof_xye-10-0.dat`
- 2) Also could the `_xye` bit be removed and `.dat’` replaced with `.xye`. e.g. e.g. XXXXXX_tof_xye-10-0.dat -> e.g. `XXXXXX_tof-10-0.xye`. This would mirror the filename behaviour of `focus_mode=“Trans”`.
-  3) At the moment, all our focussed files go into a single folder. It would be nice if we could separate these into sub-folders named by file type. I’m envisaging folders for ‘ToF’, ‘d’, and ‘GSAS’ and then the .nxs files would continue to just go into the current directory as normal. I think it’s done on some of the other instruments.

DannyHindson, 2021-03-30:
Also look at reproducing this crash. Daniel had a go on Ubuntu and didn't see it:
I think Chris originally mentioned cycle 19_1 as being problematic due to the HRPD cards/absorption correction. Though I think that problem has now gone away for the standard frame, I’m unable to create_vanadium for long_mode=True. Tt modes 77 and 88 both crash, and tt35 runs to completion but no data are shown. I don’t seem to have this problem in other cycles.

DannyHindson, 2022-08-09:
Script to run a PEARL reduction. The `C:\PEARL\` folder will need changing to wherever your config files are stored:
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

from isis_powder.pearl import Pearl

pearl_focus = Pearl(user_name="Danny",
                    output_directory=r"C:\PEARL",   #directory above cycle number 
                    config_file=r"C:\PEARL\User_experiment.yaml",
                    calibration_directory=r"C:\PEARL\calib_critical_files",
                    calibration_mapping_file=r"C:\PEARL\calib_critical_files\pearl.yaml")

pearl_focus.create_vanadium(run_in_cycle=116374, do_absorb_corrections=True)

pearl_focus.focus(run_number="116374", file_ext=".s02", perform_attenuation=False, 
                  subtract_empty_instrument=False)
```
In addition to the script above you'll also need a zip file with some config (.yaml) files in it and a couple of data files (eg a .nxs file). This is too big to attach to this issue so I'll put it somewhere

DannyHindson, 2022-08-09:
Running this script I get the following output files produced from the pearl.focus line. This makes me think that item ii) has already been done:
[image]

closed 2023-04-20 by thomashampson