mantidproject/mantid #30887: Cleanup IPeak interface

jmborr, 2021-03-14:
**TODO:**  
- [ ] remove `getDetector`
- [ ] remove `getDetectorPosition`, `getDetectorPositionNoCheck`
- [ ] remove `getBankName` from *Ipeak*; move `setBankName` from *BasePeak* to *Peak*. move attribute `bankName` from *BasePeak* to *Peak*
- [ ] remove `getRow`, `setRow`, `getCol`, `setCol`; move attributes `row` and `col` from *BasePeak* to *Peak*

```
IPeak 
   <-- MockIPeak
   <-- BasePeak
          <-- LeanElasticPeak
          <-- Peak
```

RichardWaiteSTFC, 2021-03-15:
The WISH people still want access to some of this information - specifically they want to be able to extract the detectorID (IPeak method `getDetectorID()`) . This will be used to determine the workspace index and then call `setMatrixWorkspace` on a `BackToBackExponential` IPeakFunction object to use the Parameters.xml info to make a good guess for A,B,S and hence the fwhm (this will be made possible in PR #30835). This will be used to make an initial guess for the integration radius (input to IntegratePeaksMD). Would this issue interfere with that proposed workflow?

jmborr, 2021-03-15:
All these methods are still defined in concrete class `DataObjects::Peak` so there should be no problem.

RichardWaiteSTFC, 2021-03-15:
> All these methods are still defined in concrete class `DataObjects::Peak` so there should be no problem.

Great thanks for confirming that. Just out of interest what is the plan for the LeanPeaks objects?

closed 2021-04-06 by jmborr