mantidproject/mantid #30889: Investigate warnings when loading old ISIS RAW data log files

martyngigg, 2021-03-15:
**Original reporter:** ISIS/Steve King[nobody]

### Expected behavior

An ISIS raw file loads successfully with the log files too.

### Actual behavior

Warniing messages are produced saying that the log files are not standard.

```
SANSLoad started
Unable to successfully run LoadLog Child Algorithm:
File\\isis\inst$\NDXLOQ\Instrument\data\cycle_02_4\LOQ96546.log is not a standard ISIS log file. Expected to be a file starting with DateTime String format.
Cannot process ICPevent log. Period 1 assumed for all data.
Unable to successfully run LoadLog Child Algorithm:
File\\isis\inst$\NDXLOQ\Instrument\data\cycle_02_4\LOQ96545.log is not a standard ISIS log file. Expected to be a file starting with DateTime String format.
Cannot process ICPevent log. Period 1 assumed for all data.
Unable to successfully run LoadLog Child Algorithm:
File\\isis\inst$\NDXLOQ\Instrument\data\cycle_02_4\LOQ96537.log is not a standard ISIS log file. Expected to be a file starting with DateTime String format.
Cannot process ICPevent log. Period 1 assumed for all data.
Unable to successfully run LoadLog Child Algorithm:
File\\isis\inst$\NDXLOQ\Instrument\data\cycle_02_4\LOQ96543.log is not a standard ISIS log file. Expected to be a file starting with DateTime String format.
Cannot process ICPevent log. Period 1 assumed for all data.
Unable to successfully run LoadLog Child Algorithm:
File\\isis\inst$\NDXLOQ\Instrument\data\cycle_02_4\LOQ96544.log is not a standard ISIS log file. Expected to be a file starting with DateTime String format.
Cannot process ICPevent log. Period 1 assumed for all data.
Unable to successfully run LoadLog Child Algorithm:
File\\isis\inst$\NDXLOQ\Instrument\data\cycle_02_4\LOQ96537.log is not a standard ISIS log file. Expected to be a file starting with DateTime String format.
Cannot process ICPevent log. Period 1 assumed for all data.
```
### Steps to reproduce the behavior
This will need the ISIS archive set up on Linux and the VPN on for both Linux and Windows.

* In workbench, click `Load` and type one of the following into the filename box:
  * Windows: `\\isis\inst$\NDXLOQ\Instrument\data\cycle_02_4\LOQ96537.raw`
  * Linux: `/archive/ndxloq/Instrument/data/cycle_02_4/LOQ96546.raw`
* Ensure `LoadLogFiles` is checked
* Click `Run` and the log will print the error messages

### Platforms affected
All

SpencerHowells, 2021-03-15:
@martyngigg We are having the same problems with IRIS data. Also the temperature value has an unknown scaling factor. Freddie couldn't find any info - nor could I!

martyngigg, 2021-03-15:
@SpencerHowells Thanks for the info. We'll see what we find when we start investigating.

Are your newer files okay? The newer LOQ ones seem fine.

SpencerHowells, 2021-03-15:
@martyngigg Newer files are ok.

thomashampson, 2021-03-18:
The LoadLog algorithm was written in 2007 after the format of the log files changed, and it only supports the newer format. It looks like we'll have to implement some new parsing algorithms to extract the data.

SpencerHowells, 2021-03-18:
@thomashampson I’ve spoken to the LOQ scientist & he only wants to get rid of the error warning and ignore the file. That’s also my opinion.

SpencerHowells, 2021-03-18:
@thomashampson Got the previous comment a bit wrong. We’ve both ignored the message & file for our purposes. In principle it probably needs to be sorted out. However we’re having trouble working out how to interpret temperature values which are called TEMP/1. Other instruments may have other variables, so formats may differ.

martyngigg, 2021-03-19:
@SpencerHowells Thanks for this. It sounds like a good first step would be then to have a good warning message saying something like "The log file looks like it is in the unsupported CMAC file format." rather than the LoadLog file just failing and then we can work out if we actually need to support loading them?

martyngigg, 2021-03-22:
Further thoughts from Spencer just to keep a record:
```
I’ve been thinking again following an email from Steve King on LOQ.
So where are we?
The Mantid load alg was written in 2007. We do not know when instruments changed the format.
The Genie programs did not read log files automatically & Freddie has not found the code to read it yet.
IRIS has temperature data & we do not know the scaling value to apply. Freddie believes this will depend on the type of temperature controller and/or how it is setup.
We do not know what other instruments use.
 
My data is from 2003, was analysed using Genie and is published. I only use it for testing programs which do not need temperature values.
Steve didn’t need the log file for LOQ either. He just thought that as a matter of principle it needs to be fixed.
 
There then seems to be 2 possible ways forward:
1 Just give a warning that the log file is unsupported
2 Read the CAMAC data ‘as is’ into Sample Logs. The user can then inspect it. If it needs to be used then the instrument specific programs would need to be modified to use it in some way.
 
Method 2 needs to be instrument independent as we don’t know at present what is needed.
Also what is the demand for it? The 2 users so far have no need for it. We should await more requests before being too involved.
After all, this is data pre-2007 & should have been analysed by now!
```

We agreed that at this point there is not enough information or need for loading the CMAC style files and we will simply issue a warning if the format look like the CMAC style and that we don't support them.

PR #30964 (2021-03-30): new warning message for loading old log files
**Description of work.**
Old log files (pre ~2007) are not supported by the LoadLog algorithm, but the error message did not report this to the user, e.g.:
```
Unable to successfully run LoadLog Child Algorithm:
File\\isis\inst$\NDXLOQ\Instrument\data\cycle_02_4\LOQ96546.log is not a standard ISIS log file. Expected to be a file starting with DateTime String format.
Cannot process ICPevent log. Period 1 assumed for all data.
```

A check has been added to determine whether the date-time format is consistent with old style log files. If the check is [... cut]
closed 2021-03-30 by gvardany