mantidproject/mantid #30891: Resize cursor for instrument viewer

thomashampson, 2021-03-15:
### Expected behavior
As a user, when manipulating a shape in the instrument view (either "pick" of "draw" tab), if I hover over the corner handles I expect the cursor to change to a `Qt::SizeFDiagCursor` or `Qt::SizeBDiagCursor`.

### Actual behavior
Currently there is no feedback from the UI to tell me I'm hovering over a corner handle. The `Qt::SizeAllCursor` is used for both moving and resizing, which can make resizing rectangles a bit cumbersome.

### Steps to reproduce the behavior
Load a workspace and open the instrument viewer. Go to the "pick" or "draw" tab and draw a rectangle and resize it.

thomashampson, 2022-11-25:
@SilkeSchomann reopening stale issue after you mentioned it would be a nice UI improvement

AnthonyLim23, 2022-11-29:
[image] [Resize cursor for instrument view](https://trello.com/c/ovn8aV1R/22-resize-cursor-for-instrument-view)

PR #34853 (2022-12-19): Added cursor feedback for Shape2DResizing
**Description of work.**

When a shape is drawn onto the Instrument Viewer, using the Pick or Draw tab, there is now cursor feedback when interacting with the shape's control points.

**To test:**

 - Load some data
 - Right click and open the Instrument Viewer
 - Go to the pick tab
   - Draw a **Rectangle** - Hover over each of the control points (corners) and check that the cursor changes to something sensible (the correct diagonal arrow head in this case)
   - Repeat this for the **Ellipse**, **Rectangular Ring**, and **Circular Sector** shapes

Fix [... cut]
closed 2022-12-19 by martyngigg