mantidproject/mantid #30908: Remove duplicate WISH system test files

DannyHindson, 2021-03-17:
Different versions of the following files exist in both the Testing\Data\SystemTest and Testing\SystemTests\tests\framework\reference:

WISH40503-5_6no_absorb_raw.nxs.md5
WISH40503-5_6raw.nxs.md5
WishAnalysis.nxs.md5

This is a bit confusing because it's possible to update the ones in the "reference" folder and they have no effect because the files in the get used at the moment. I think the files should be in the Testing\SystemTests\tests\framework\reference because they're used in the validate method of the WISH system tests. 

So the actions required are to delete the files from Testing\Data\SystemTest and bring the files in the reference folder up to date

DannyHindson, 2021-03-25:
@RichardWaiteSTFC - while doing my big ConvertUnits change I came across a few things in the system tests that caused by a few problems and need tidying up. I'm about to tidy up this issue which is that there are some duplicate WISH files in GitHub for the system tests. Are you happy for me to go ahead remove these duplicates?

RichardWaiteSTFC, 2021-03-25:
> @RichardWaiteSTFC - while doing my big ConvertUnits change I came across a few things in the system tests that caused by a few problems and need tidying up. I'm about to tidy up this issue which is [...]

Yes of course, thanks for doing that. I wasn't ware there were two folders - so in what scenario would one put a file in Testing\Data\SystemTest rather than Testing\SystemTests\tests\framework\reference?

DannyHindson, 2021-03-25:
> Yes of course, thanks for doing that. I wasn't ware there were two folders - so in what scenario would one put a file in Testing\Data\SystemTest rather than [...]

Well I've not read this in our documentation anywhere so this is my own take - but it seems to be the case that the files that are checked at the end of a system test in the validate method are stored in Testing\SystemTests\tests\framework\reference and any other files such as those used as inputs to a test go in Testing\Data\SystemTest

PR #30975 (2021-03-26): Remove duplicate WISH system test files
**Description of work.**

Different versions of the following files existed in both the Testing\Data\SystemTest and Testing\SystemTests\tests\framework\reference:

WISH40503-5_6no_absorb_raw.nxs.md5
WISH40503-5_6raw.nxs.md5
WishAnalysis.nxs.md5

The files in the Testing\Data folder seemed to always get found first by Mantid during the system tests. This was a bit confusing because it was possible to update the ones in the "reference" folder without having any effect. 

The files should only exist in the Testing\SystemTests\tests\framework\reference becau [... cut]
closed 2021-03-26 by martyngigg