mantidproject/mantid #30912: Mantid fails to find Qt6

quantumsteve, 2021-03-18:
### Expected behavior

Build Mantid on MacOS with Qt and PyQt provided by homebrew

### Actual behavior

```
CMake Error at CMakeLists.txt:109 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/Users/svh/Documents/MantidProject/build/CMakeFiles/CMakeOutput.log".
```

### Steps to reproduce the behavior

### Platforms affected

Currently MacOS with up-to-date homebrew (3.0.7-19-g3c3bf1c)

With [Qt's LTS versions being commercial-only](https://www.phoronix.com/scan.php?page=news_item&px=Qt-Going-More-Commercial), expect Linux distributions to drop Qt5 faster than they did Qt4.

quantumsteve, 2021-03-18:
One can try to make use of the keg-only qt@5 and pyqt@5 formulas. Longer-term, I recommend adding support for Qt6 and PyQt6.

One potential issue: [QtPy](https://github.com/spyder-ide/qtpy) doesn't currently have support to PyQt6 or PySide6.

A [PR adding PySide6 suport](https://github.com/spyder-ide/qtpy/pull/225) is under review.

quantumsteve, 2021-03-26:
https://youtu.be/gbZZEIAXt3c

martyngigg, 2021-03-29:
Thanks for pointing this out. The point about moving to 5.15 first is an interesting one to consider.

I've made a simple fix so we can find Qt5 in the new homebrew path but we now have a problem that `sip` has move to v6 with a completely new build system that doesn't work for us so I'm trying to find a workaround for now.

DavidFair, 2021-04-07:
Just ran into this today, I found that I can get a bit further with:
`export Qt5_DIR=/usr/local/opt/qt5/lib/cmake/Qt5` 

But then we fall over on PyQt5 instead, regardless of trying both pip and brew install

martyngigg, 2021-04-08:
Yes the Qt 5 stuff is pretty trivial at this point. The main issue is PyQt5/sip is now built with sip v6 that has a completely new build system. I'm able to build locally if I reset the homebrew-core tap in `/usr/local/Homebrew/Taps/homebrew/homebrew-core` to commit `5048c9cf11c4a74a654d80dc158cae5f072850fa` which is the last line the old version of sip was used.

This has also moved to python 3.9 now so the pip requirements from our homebrew repo might need to be reinstalled.

PR #31086 (2021-04-22): Add support for sip-build to CMake
**Description of work.**

Updates our cmake support for both versions of the sip build system. This will allow us to build with the HEAD of brew but also support the other operating systems.

**To test:**

* Run `brew upgrade`
  * This will bring down Python 3.9. Check this is what you see when the install finishes. You might need to restart the shell if not.
  * Run `python3 -m pip install -U -r /usr/local/Homebrew/Library/Taps/mantidproject/homebrew-mantid/requirements.txt` to reinstall the Python requirements.
* Try a clean build of mantid.

Fixes #30912 
 [... cut]
closed 2021-04-22 by SimonHeybrock