mantidproject/mantid #30992: Make FitOptionsBrowser more flexible

robertapplin, 2021-03-29:
### Description
This issue is to refactor the Fit Options Browser to make it more flexible and maintainable. A list of tasks we need to complete are:
- [ ] Refactor `FitOptionsBrowser` to make it more maintainable and add tests (the `BasicFitOptionsBrowser` and its tests can be a starting point for this)
- [ ] Add a method to `FitOptionsBrowser` that allows certain properties to be hidden.
- [ ] Add a method to `FitOptionsBrowser` that can be used to hide sub-properties of the `Minimizer` property
- [ ] Remove the `BasicFitOptionsBrowser`. This was probably not the best solution to allow flexibility of properties.

Then:
- [ ] Decide on a plan to replace `FitPropertyBrowser` with something more maintainable. Possibly the `FitOptionsBrowser` can be the replacement

martyngigg, 2021-03-30:
Can I ask that we come up with and agree a design for this going forward before we write any code? I'm concerned that we want to plan something so we can phase out the older `FitPropertyBrowser` and replace it with something more configurable and maintainable but we need to make sure it can work everywhere we use it.

robertapplin, 2021-03-30:
Yes I will look into how we might want to design this. I might just need to do a bit of discovery of what the `FitOptionsBrowser` can currently do because Roman already replaced the `FitPropertyBrowser` used in indirect with a FunctionBrowser and FitOptionsBrowser so I would assume it already does most of the things we need it to do. But I will do some discovery and see what comes out of that

robertapplin, 2021-07-02:
It looks like I will add some options to the `BasicFitOptionsBrowser` which are not just fit options but also options for what to include in a fit script (for instance the plotting option described here #31955 ). So the `BasicFitOptionsBrowser` will probably be renamed to `FitScriptOptionsBrowser`. In which case the need to make the `FitOptionsBrowser` more flexible might be postponed as its not immediately required

robertapplin, 2021-08-04:
Closed because the `BasicFitOptionsBrowser` has been renamed and diverged in nature to be the `FitScriptOptionsBrowser` in #31974

closed 2021-08-04 by robertapplin