mantidproject/mantid #31000: Mismatched system test data found when fixing the column name error in CompareWorkspaces for Peak Workspace

KedoKudo, 2021-03-30:
> - Original Gitlab task: [SCD271](https://code.ornl.gov/sns-hfir-scse/diffraction/single-crystal/single-crystal-diffraction/-/issues/271)
> - Related branch:   [SCD271_FixColNameErrorInCompareWorkspacesPWS](https://github.com/mantidproject/mantid/tree/SCD271_FixColNameErrorInCompareWorkspacesPWS) 

Several system tests are failing after fixing a bug where the column name was incorrectly set for Peak Workspace in `CompareWorkspaces`, including

## [ReduceOneSCD_Run](https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/framework/ReduceOneSCD_Run.py)

The error is related to the set tolerance lower than the difference
```
CompareWorkspaces-[Notice] Wavelength
CompareWorkspaces-[Notice] s1 = 0.740069
CompareWorkspaces-[Notice] s2 = 0.740069
CompareWorkspaces-[Notice] std::fabs(s1 - s2) = 1.27904e-07
CompareWorkspaces-[Notice] tolerance = 1e-08
CompareWorkspaces-[Notice] Data mismatch at cell (row#,col#): (0,5)
CompareWorkspaces-[Notice] The workspaces did not match: Data mismatch
CompareWorkspaces-[Notice] CompareWorkspaces successful, Duration 0.00 seconds
```

## [Diffraction_Workflow_Test](https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/framework/Diffraction_Workflow_Test.py)

The results cached in reference file is different from calculated value (reasons unknown)
```
CompareWorkspaces-[Notice] DSpacing
CompareWorkspaces-[Notice] s1 = 0.701967
CompareWorkspaces-[Notice] s2 = 0.702038
CompareWorkspaces-[Notice] std::fabs(s1 - s2) = 7.09542e-05
CompareWorkspaces-[Notice] tolerance = 1e-08
CompareWorkspaces-[Notice] Data mismatch at cell (row#,col#): (0,8)
CompareWorkspaces-[Notice] The workspaces did not match: Data mismatch
CompareWorkspaces-[Notice] CompareWorkspaces successful, Duration 0.00 seconds
```

## [FindSatellitePeaksTest](https://github.com/mantidproject/mantid/blob/master/Testing/SystemTests/tests/framework/FindSatellitePeaksTest.py)
- FindSatellitePeaksTestFixedNumQ
  The reference file contains a very different number than the one calculated from the system test
```
CompareWorkspaces-[Notice] BinCount
CompareWorkspaces-[Notice] s1 = 60818
CompareWorkspaces-[Notice] s2 = 0
CompareWorkspaces-[Notice] std::fabs(s1 - s2) = 60818
CompareWorkspaces-[Notice] tolerance = 1e-08
CompareWorkspaces-[Notice] Data mismatch at cell (row#,col#): (0,12)
CompareWorkspaces-[Notice] The workspaces did not match: Data mismatch
CompareWorkspaces-[Notice] CompareWorkspaces successful, Duration 0.00 seconds
``` 
- FindSatellitePeaksTestAutoFindQ
  Similar issue as the previous one, the reference file seems to contains a very different data set
```
CompareWorkspaces-[Notice] BinCount
CompareWorkspaces-[Notice] s1 = 60818
CompareWorkspaces-[Notice] s2 = 0
CompareWorkspaces-[Notice] std::fabs(s1 - s2) = 60818
CompareWorkspaces-[Notice] tolerance = 1e-08
CompareWorkspaces-[Notice] Data mismatch at cell (row#,col#): (0,12)
CompareWorkspaces-[Notice] The workspaces did not match: Data mismatch
CompareWorkspaces-[Notice] CompareWorkspaces successful, Duration 0.00 seconds
```

KedoKudo, 2021-03-30:
@robertapplin Could you take a look at these affected system tests?  Thanks :pray:

robertapplin, 2021-03-30:
Hi @KedoKudo , I don't know much about this area of Mantid in the tests - I have only moved these tests to a different folder location a few months ago. I am assuming #30001 is a fix for this?

KedoKudo, 2021-03-30:
> Hi @KedoKudo , I don't know much about this area of Mantid in the tests - I have only moved these tests to a different folder location a few months ago. I am assuming #30001 is a fix for this? 

Actually, #30001 fixes a bug in CompareWorkspaces and the PR cannot be merged because of the broken system tests listed here.  Do you happen to know who I should contact to fix these system tests?

robertapplin, 2021-03-30:
This looks like single crystal, so maybe @RichardWaiteSTFC will know more about these tests?

RichardWaiteSTFC, 2021-03-30:
Looks like it's just FindSatellitePeaksTest that is broken so it's probably worth checking that algorithm is working as expected - if not maybe it's a separate issue. The tolerance in Diffraction_Workflow_Test and ReduceOneSCD_Run looks va bit small (e.g. limit of WISH resolution at backscattering is 0.05% on d-spacing at backscattering) so probably just accumulation of smaller changes over time and they are still doing what we expect.

KedoKudo, 2021-03-31:
Did some additional investigation and here are the findings:

ReduceOneSCD_Run
-------------------------------
Raising the tolerance of system test help it pass.

Diffraction_Workflow_Test
-----------------------------------
Raising the tolerance of the system test help it pass the `DSpacing` comparison, but now we are getting another column that has mismatched data
```
CompareWorkspaces-[Notice] Intens
CompareWorkspaces-[Notice] s1 = 9270.48
CompareWorkspaces-[Notice] s2 = 9254.55
CompareWorkspaces-[Notice] std::fabs(s1 - s2) = 15.93
CompareWorkspaces-[Notice] tolerance = 0.0001
CompareWorkspaces-[Notice] Data mismatch at cell (row#,col#): (0,9)
CompareWorkspaces-[Notice] The workspaces did not match: Data mismatch
CompareWorkspaces-[Notice] CompareWorkspaces successful, Duration 0.00 seconds
```
It seems like we need to update the reference file as it is most likely outdated.  According to the git history, @martyngigg is the original author, and I believe the safest way to fix this system test is to let the original author update the test reference file (if possible). 
Alternatively, we could also disable this test if the associated algorithms are no longer on active duty.

FindSatellitePeaksTest
-------------------------------
The associated algorithm, `FindSatellitePeaks` is still functional based on a test using the testing script provided on the documentation page.  However, we still have the issue where the reference data does not seem to match the system test at all. @RichardWaiteSTFC , do you happen to know if anybody is still actively using `FindSatellitePeaks` in their workflow?  If not, we might be better off remove this system test all together.

RichardWaiteSTFC, 2021-03-31:
The `Diffraction_Workflow_Test ` uses` IntegratePeaksMD` to integrate the peaks - there have been a few recent changes to this algorithm - but none that should effect the intensity of a spherical region. However, it looks like given the d-spacing of the peak shifted a little, is it possible a change to CentroidPeaksMD means we're integrating over a slightly different region? I think you're recent work on the box checking means we can be pretty confident the IntegratePeaksMD is working.

martyngigg, 2021-03-31:
I'm afraid I can't probably help much here. I look like the author of the `Diffraction_Workflow_Test` because I was the one who batch committed a lot of tests that used to be in a separate repository. If the new answer is acceptable then it does sound like just an update to the reference result would be okay?

RichardWaiteSTFC, 2021-03-31:
I think the new answer is not terrible - the difference is probably smaller than the error (assuming TOPAZ data is event the error ~100) so it's just a case of trying to understand how the difference came about - if we can find a reasonable cause I think it's OK to update the reference data.

KedoKudo, 2021-03-31:
> I'm afraid I can't probably help much here. I look like the author of the `Diffraction_Workflow_Test` because I was the one who batch committed a lot of tests that used to be in a separate [...]

It sounds like the solution for `Diffraction_Workflow_Test` is just an update of the reference file, which I will do in the related PR.
What should we do about `FindSatellitePeaksTest`? @RichardWaiteSTFC @martyngigg

PR #31001 (2021-04-06): Fix column name error in CompareWorkspaces for Peak Workspace
**Description of work.**

- Original Gitlab task: [SCD271](https://code.ornl.gov/sns-hfir-scse/diffraction/single-crystal/single-crystal-diffraction/-/issues/271)
- This is #31002 into `master`

The column names of peak workspace defined in `ComapreWorkspaces` are all in lower cases whereas the current Mantid is using Camel cased column names for Peak Workspaces.
This PR fix this issue and will enable `CompareWorkspaces` to check  the content of two input Peak Workpaces instead of only checking the total number peaks.

**To test:**
 [... cut]
closed 2021-04-06 by peterfpeterson