mantidproject/mantid #31048: Improvements to memory widget

peterfpeterson, 2021-04-05:
The memory widget introduced in #30929 is a great improvement for usability. What would make it even better, is if the memory bar included a section for memory used by the current process as well. The way to find the information is by using
```python
import os
import psutil
import pwd

CONVERSION_FACTOR_BYTES_TO_GB = 1.0 / (1024 * 1024 * 1024)

pid = os.getpid()
username = pwd.getpwuid(os.getuid()).pw_name
found = None
for proc in psutil.process_iter(): 
    pinfo = proc.as_dict(attrs=['pid', 'name', 'username']) 
    if pinfo['username'] == username and pinfo['pid'] != pid: 
        pinfo['vms_GiB'] = proc.memory_info().vms * CONVERSION_FACTOR_BYTES_TO_GB
        found = pinfo        
        break

print(found)
```

Which would be added to `qt/python/mantidqt/widgets/memorywidget/memoryinfo.py`. Then the issue to solve is creating a "progress bar" type widget with 3 colors (used by process, rest that is used, free).

It would also be clearer to read client code for the `get_memory_info` object to return a `dict` or `named_tuple`.

KedoKudo, 2021-04-06:
I would recommend adding an `advanced` option where the memory usage of each workspace can be visualized.
This will help user have a better understanding how the analysis data service is managing the memory usage in the backend.

peterfpeterson, 2022-03-29:
This was fixed by #33665

closed 2022-03-29 by peterfpeterson