mantidproject/mantid #31070: Crash when closing a matplotlib subplot for the second time

robertapplin, 2021-04-07:
### Expected behavior
There should be no crash, the subplot window should close.

### Actual behavior
An error message appears, and a possible crash.
[image]

### Steps to reproduce the behavior
1. Run the script below twice
2. Close both plots that are opened. The crash happens

```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

ws = CreateSampleWorkspace()

fig, axes = plt.subplots(1, 5)
for j in range(5):
    last_index = len(ws.dataY(j))
    axes[j].set_title(ws.name())
    axes[j].plot(ws.dataX(j)[:last_index], ws.dataY(j), label="1")
    axes[j].plot(ws.dataX(j+1)[:last_index], ws.dataY(j+1), label="2")
    axes[j].plot(ws.dataX(j+2)[:last_index], ws.dataY(j+2), label="3")
    axes[j].legend()

plt.show()
```

### Platforms affected
All, found on windows

DanielMurphy22, 2021-04-08:
This may be the same as https://github.com/mantidproject/mantid/issues/31031

robertapplin, 2021-04-08:
Closed because is a duplicate of #31031

closed 2021-04-08 by robertapplin