mantidproject/mantid #31073: New YAML output format for SaveReflectometryASCII

gemmaguest, 2021-04-07:
**Original reporter**: Max at ISIS

ISIS Reflectometry have a new YAML based ASCII format they would like to be able to save reduced data to. We already support multiple ASCII format using the `SaveReflectometryASCII` algorithm, so this would be added as another option to that algorithm, and exposed on the `Save ASCII` tab on the ISIS Reflectometry interface.

The format is simply something like the following (see also https://www.reflectometry.org/working_groups/file_formats/examples):

```
# creator:
#   name        : Jochen Stahn
#   affiliation : PSI
#   time        : 2020/04/06/13:21:18
# data source:
#   experiment:
#     probe              : neutrons
#     sample:
#       name             : Ni1000
# reduction:
#   software: eos.py
#     call : eos.py -Y 2020 -n 1925-1927 -y 9,55 ni1000 -O -0.2 -r 1064 -s 1 -i -a 0.005 -e
# data:
#   column 1 : Qz / Aa^-1
#   column 2 : RQz
#   column 3 : sigma RQz , standard deviation
#            1               2               3
1.03563296e-02  3.88100068e+00  4.33909068e+00
1.06717294e-02  1.16430511e+01  8.89252719e+00
```

From Max:
- I think most entries can be transferred easily, but I see some potential confusion with "sample name" as opposed to " sample or run title". I will raise this at the next ORSO meeting.
- Regarding the "reduction call", we could list the most top level call, e.g. ReflectometryISISLoadAndProcess with all the specified parameters.
- As you can see on the website, we have also discussed a more comprehensive file format, but I think for now we should aim for the simpler minimal version.

DavidFair, 2021-08-23:
@gemmaguest I've untagged this from 6.2, but I thought I'd ping you if you want it to target 6.3?

gemmaguest, 2022-11-17:
It would be good if we could add support at least a minimal YAML output format as soon as possible so that ISIS can shape development of these output formats

jfkcooper, 2023-05-09:
After working on this a little and discussing with Rachel we found that there are several blockers to implementing the full standard. 
A file saving algorithm (in my view at least) should only need to have access to the workspace which it is trying to save, which means that any metadata which is required by the standard should somehow be in the workspace. There are a reasonable number of entries for which this is true and easily accessible:
- instrument
- start date
- probe
- facility
- proposal ID (RB number)
- data DOI
- sample name
- reduction software name and version
- Q values
- Reflectivity values
- Error on reflectivity values

There are however a number of required entries which are not so easy to extract from the workspace (though do in principle exist in them). These are all present in the workspace history, but this is not a good place to be getting the values from as it is an unintended use case and would make the algorithm very fragile to changes in the format of the history:
- Resolution
- Incident angles which were measured at
- Data files which contributed to the reduced and stitched workspace
- Wavelength ranges used for each of the angles

In addition, there are required entries which (again as far as I can tell) do not exist in the mantid workspaces, but do exist in the nexus files from which the data was loaded:
- Data owner names
- Data owner affiliations

Finally, there are the required entries which do not exist anywhere. These are not saved in the nexus files (and really should be), with the treatment of the subsequent files done solely on what is "known". Note that these require changes on the IBEX side:
- Polarisation (if the measurement was a polarised measurement or not)
- If polarised, then what spin state each of the data files is (up, down,  up-up, up-down, down-up, or down-down)
- Slightly less possible in an automated way, but - whether the dataset corresponds to an: air/solid, air/liquid, liquid/liquid, liquid/solid, solid/liquid, or solid air system (which is information known at the time of running, but I don't believe has a prescribed way of recording it into the nexus files)

After the discussion with Rachel, we believe that the best course of action is to try to proceed without any of the properties listed above *except* the resolution, which is required by the analysis programs. This would be supplied as an additional input to the algorithm (in addition to the workspace name and output filepath). Since the algorithm is unlikely to be used by itself, we would also try to work on getting the reflectometry GUI to be able to auto-populate this when saving datafiles.

This would put the algorithm in a state where it technically does not conform to the standard, but is still considerably better than not being able to save into the ORSO format at all, and is preferable to having a large amount of hacky code which would need replacing at a later date once the logging issues detailed above have been sorted. 

Lastly, there is also the issue of if and how to include `orsopy` as a dependency. There is some work which may happen on their side to create a condaforge distribution in additoin to their PyPi version, which may help here, though this is not confirmed and still leaves the question of *if* it should be included.

jfkcooper, 2023-06-08:
[SaveReflectometryORSO.txt](https://github.com/mantidproject/mantid/files/11687079/SaveReflectometryORSO.txt)
[SaveReflectometryORSOTest.txt](https://github.com/mantidproject/mantid/files/11687080/SaveReflectometryORSOTest.txt)
[test_ort_file.txt](https://github.com/mantidproject/mantid/files/11687081/test_ort_file.txt)
I attach the first version of the writer, an example output, and unit tests for this. The writer is annotated with pieces which are or may be ISIS specific

rbauststfc, 2023-06-15:
For when we pick this up:

rbauststfc, 2023-07-26:
Background info for POLREF from Andrew:

> On POLREF, for specular reflectivity (for off-specular the output is a bit different) we basically only save out a 3-column ASCII with no header...the 3 columns are Q, R, dR. We have each individual spin state as a different file.
> 
> In future, we (POLREF) would like to go a similar way to the NIST NCNR orso file, where we include all spin states in one file - we would be looking to have 8 columns: Q, R, dR, dQ, lambda, dlambda, incident theta, d incident theta. For POLREF at least, initially we would work with unstitched data, but we might append the stitched data to the end of the file - I haven't worked this out in my head yet.

The various options/data sets are why Reflectometry are so interested in the ORSO Nexus file.

jfkcooper, 2023-07-26:
As it stands, I believe only the ascii data format has been accepted by the ORSO organisation, and the HDF/Nexus style is still subject to potentially a reasonable amount of change (depending on if ORSO chooses to got for a Nexus standard, or just using the HDF5 format). This may be relevant in the future, but I would stay away from it for now in mantid.

rbauststfc, 2023-07-26:
> As it stands, I believe only the ascii data format has been accepted by the ORSO organisation, and the HDF/Nexus style is still subject to potentially a reasonable amount of change (depending on if [...]

Thanks @jfkcooper, this is just capturing information for the future.

maxskoda, 2023-08-02:
Hi Rachel,

As discussed, here are some comments regarding the test .ort file you have shared:

1. We should really change the first line in the header, but it seems this is hard coded in osropy.file.orso:
ORSO_VERSION = "1.0"
ORSO_DESIGNATE = (
    f"# ORSO reflectivity data file | {ORSO_VERSION} standard " "| YAML encoding | https://www.reflectometry.org/" 
)
I will talk to the orsopy coders about this. In the meantime we could just add a comment in the save_orso method. I think this will show up just below the first line. It could read sth like: "Mantid@ISIS output may not be fully ORSO compliant" 

2. start_date issue: I'm not sure what the timestamp refers to - the beginning of the whole experiment, the beginning of the data set or what. Since the .ort file will most likely contain stitched and normalised data, the start date/time doesn't exactly make sense here. Again, I will check with ORSO.

3.  data_files: I believe this should be a list of the individual angle files, in full format, as already noted

4. reduction: it would be good to use the "corrections" parameter in orsopy.fileio.reduction.Reduction to indicate at the very least the transmission, calibration and flood files that have been used for the reduction. Ideally one could also include information such as "normalised by monitor integral"

5.a) reduction - timestamp: Yes, some approximate time stamp of when the reduced workspace was created (as opposed to when the .ort file was created).
b) creator: orsopy says: "creator (Optional[[Person] https://orsopy.readthedocs.io/en/latest/orsopy.fileio.base.html#orsopy.fileio.base.Person)]) – The person or routine who created the reduced file." I think this could be one or more of: [Mantid->RROA or similar algorithm name, local contact, entire experimental team]

6. sQz - as note before, this needs to be (slit resolution) * (Qz value), e.g.: 0.04*4.7711570593988312e-03

maxskoda, 2023-08-02:
> As it stands, I believe only the ascii data format has been accepted by the ORSO organisation, and the HDF/Nexus style is still subject to potentially a reasonable amount of change (depending on if [...]

While I agree that the binary/HDf5 .orb format is still in a state of flux, I think that this the time to start thinking about how all the necessary information can be collected. This could inform the structure of the .orb file and avoid the situation that we find ourselves in now with the .ort format, where we have the orsopy library, but we cannot fully use it due to missing meta-information.

rbauststfc, 2023-08-30:
Summary of how we expect this to develop in the future:

- It should be possible to include both stitched and unstitched (individual angle) data in a single output file.
- Stitched data would have 4 columns (the first 4 specified by the ORSO format, however if resolution information isn't available then we shouldn't include the fourth column).
- Unstitched data on POLREF would have the following additional 4 columns: `lambda`, `dlambda`, `incident theta`, `d incident theta` (we don't have a calculation for `d incident theta` yet, so it can't currently be implemented). Note, for this 8 column version we would need to have a column of `nan` values if resolution wasn't available for some reason.
- POLREF would like to have all the spin states saved into a single file (at the moment, individual files are output for each spin state).
- To be confirmed, but there may be some difference in which header fields the different ISIS instruments want to include. We expect the same header fields to be included (when info is available) for all experiments on the same instrument, though.

rbauststfc, 2023-09-01:
Information about resolution calculation:

- The existing ASCII save formats use calculation dQ/Q = x[1] - x[0]/x[1] for the resolution, as it was previously not possible to get this from the GUI. We should make the resolution calculation consistent across all Reflectometry ASCII save formats, using whatever we put into the ORSO format for all. This may be best done under a separate issue.

- Normally the dQ/Q resolution would be expected to be the same for all angles. For saving stitched data, we should be taking the value that was passed in as the stitch resolution parameter to `Stitch1DMany` as that will give the correct resolution for the data. If the data is unstitched the we can take the individual row value. We should have a parameter to specify whether or not to include the resolution data (hooked up to the GUI tick-box) so that users have flexibility over whether to output this. 

- If the resolution is not passed in and we're trying to calculate it for stitched data then we cannot use `NRCalculateSlitResolution` because the data contains multiple angles. We could use the same calculation as we're currently using for the resolution columns in the existing save output, which does not require the theta value. However, the disadvantage is that we're not providing anything that couldn't have been calculated by the fitting software anyway (as it simply depends on the data values passed in). We should have a think about if there is any other useful alternative we can offer in this situation.

closed 2023-12-11 by SilkeSchomann