mantidproject/mantid #31088: Single crystal peak integration

zjmorgan, 2021-04-08:
[SCD287](https://code.ornl.gov/sns-hfir-scse/diffraction/single-crystal/single-crystal-diffraction/-/issues/287)

Explore ways of doing single crystal peak integration with background subtraction. When integrating peaks, problems often arise when estimating the background running into powder lines or nearby satellite reflections. This is a common problem across various instruments and/or facilities, so the purpose of this issue is to have appropriate conversations on how to best approach these problems. Description will be updated...

RichardWaiteSTFC, 2021-04-08:
So there might be an easy-win here that will help a common problem. In the majority of cases when a peak region is contaminated by a powder line we don't necessarily want to integrate that peak/pass it to the refinement software. Obviously if the peak is very strong maybe you don't care, but typically the intensity and background will not be reliable (neither will the covariance estimation etc.). Given that users typically mount their crystals on one of Al Cu, CuBe, Ti - for which we know the d-spacing (and therefore |Q|) of the powder peaks, I think it would be good to give the users the option of screen/remove/integrate separately (for easy inspection) the peaks for which (Q_SX - Q_powder) < radius or bg_outer - maybe we add some fudge/scale factor to account for the fact the powder peaks have a finite width (e.g. specify some delta d/d etc. although this will change with tth). The case of satellite peaks is a bit harder - I think ultimately if the modulation wavevector is very small and it is contaminated by the Bragg peak the best option is some form of fitting a 3D profile to both parent and satellite peak together - I don't think we will be able to automate this in an algorithm like IntegratePEaksMD.

KedoKudo, 2021-04-08:
> So there might be an easy-win here that will help a common problem. In the majority of cases when a peak region is contaminated by a powder line we don't necessarily want to integrate that peak/pass it to the refinement software. Obviously if the peak is very strong maybe you don't care, but typically the intensity and background will not be reliable (neither will the covariance estimation etc.). Given that users typically mount their crystals on one of Al Cu, CuBe, Ti - for which we know the d-spacing (and therefore |Q|) of the powder peaks, I think it would be good to give the users the option (in the workflow, not in the integration algorithms) to screen/remove/integrate separately (for easy inspection) the peaks for which` (|Q_SX| - |Q_powder|) < radius` or `bg_outer` - maybe we add some fudge/scale factor to account for the fact the powder peaks have a finite width (e.g. specify some delta d/d etc. although this will change with tth). The case of satellite peaks is a bit harder - I think ultimately if the modulation wavevector is very small and it is contaminated by the Bragg peak the best option is some form of fitting a 3D profile to both parent and satellite peak together - I don't think we will be able to automate this in an algorithm like IntegratePEaksMD.

@zjmorgan Could you pass this suggestion above (pre-screening peaks by d-spacing before integration) to the TOPAZ team?  I think this could reduce the amount of work that a user need to do after the integration.

zjmorgan, 2021-04-08:
@KedoKudo will do.

closed 2021-10-13 by stale[bot]