mantidproject/mantid #31107: Manual Testing Documentation

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

In the build directory run `cmake docs-html` 
Check in the `docs/html` directoy:

* Algorithm, fit, concept and api pages should be generated
* Algorithm dialog snapshots should appear on algorithm pages in offline help
* Math formulae should appear on algorithm pages in offline help
* workflow diagrams should appear on algorithm pages in offline help

You may wish to use this script: [OpenMostDocumentationForTesting.py](https://github.com/mantidproject/mantid/blob/master/tools/scripts/OpenMostDocumentationForTesting.py) to open all the online docs pages!

Pasarus, 2021-04-28:
There is a bunch of Deprecated documentation and algorithms that have been deprecated for years:
- AddNote
- AlignDetectors
- CheckWorkspacesMatch
- ConvertEmptyToTof
- FilterEventsByLogValuePreNexus v2
- FindUBUsingMinMaxD
- RecordPythonScript
- SaveDiffFittingAscii

There was also a weird document called:
- DNSMergeRuns

Pasarus, 2021-04-28:
Issue made for this, documentation is good for release: #31258

closed 2021-04-28 by Pasarus