mantidproject/mantid #31111: Manual Testing Workbench Group 4

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/Core/Core.html

sf1919, 2021-04-27:
Sliceviewer testing part of Workbench Group 4

Tried to follow SliceViewer Testing (https://developer.mantidproject.org/Testing/SliceViewer/SliceViewer.html#sliceviewer-testing) but ran into difficulties. This will need further testing by someone else. In the meantime these are the issues I did spot with SliceViewer: -

1) On opening SliceViewer the toolbar and other elements on the GUI appear tiny. The below screenshot is what I see when the GUI is fullscreen

[image]

2) For the 2D, 3D, 4D and non-orthogonal workspaces the dropdown menus for choosing the colormap and the scale are narrow and so some of the options appear as letters and dots (e.g. tw...ed) which makes it hard to know what you are selecting. If you set the scale to Power it becomes even more narrow to accommodate an additional text box and at that point the options in the menu just appear as dots.
[image]

3) The labelling on the scale doesn't always show. Sometime's it's cropped so that you only see the first couple of digits. 
[image]

4) If toggle cursor is on sometimes the size of the Time of Flight/Spectrum table changes on a 2D plot as you move the cursor across the plot. Sometimes it will take up most of the width of the GUI (see screenshot with 1) above), other times it takes up half the width and is located to the right hand side of the screen (see below). I found this in particular when I loaded 164198.nxs from the training dataset to look at.

[image]

5) Peak overlay testing - I followed the testing document to create the peak overlays. However the file used to create them is a 2D workspace and 2D workspaces don't allow peak overlays (button greyed out). Tried using the overlays on 3D and 4D workspaces created as per testing document but generated an error (reported via Mantid).

6) It was difficult to know how to use SliceViewer and what was and wasn't expected behaviour. The Testing documentation assumed familiarity with Slice Viewer and the help in Mantid only took you to a page discussing new features. So it wasn't necessarily clear to a new user how to do some of what was being asked. Would recommend improved documentation for users to access via Help.

sf1919, 2021-04-28:
Smaller Tasks

Have now completed the Smaller Tasks for Workbench Group 4 testing and have noted the following.

1) Help continues to be difficult to use on my display. Already raised in #29979 .

2) Testing Project Load/Save - If you save a project with a plot where you have changed the Tick Format to Scientific Format but do not check the Log box then this doesn't appear to save. Opening the project the plot will have reverted to Decimal Format. If you do check the Log checkbox then it will save properly. See notes on Scripting Plots below also.

3) Manipulating Plots - When manipulating plots if you change the Tick Format to Scientific Format and do not check the Log box then only some of the ticks are visible. In example below I made the changes on the y-axis.
[image]

4) Scripting plots - If you generate a plot and change the Tick Format to Scientific Format and do not check the Log box then when you copy the script to clipboard to reuse it the resulting plot will default back to Decimal Format. The same happens when you save the script to file and open it and run it. See notes on Testing Project Load/Save above.

5) Scripting Plots - Test notes suggest using https://docs.mantidproject.org/nightly/plotting/index.html#plotting . This includes instructions specific to MantidPlot. Should they be removed?

6) Script Editor/Interpreter - Another instance where the GUI doesn't display well on my screen. When selecting the Find/Replace option the resulting toolbox is difficult to use. Buttons have partial text because they are too small for the full word/phrase and labels are also cut off. Plenty of room to space everything across the screen but all appears very bunched up to the left.

[image]

sf1919, 2021-04-28:
Larger tasks

1) Error Handling - when it can't find a file (either because the path doesn't exist or the file name is wrong) I don't find the error makes that clear. At present the error returned is
NameError: name 'name' is not defined
`File "<string>", line 7, in <module>
File "C:\MantidNightlyInstall\bin\mantid\simpleapi.py", line 141, in Load
msg = 'Problem setting "Filename" in {}-v{}: {}'.format(name, algm.name(), algm.version(), - `

2) Other - noticed there is no ? button on SliceViewer GUI.

closed 2021-04-29 by sf1919