mantidproject/mantid #31112: Manual Testing Muon

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See instructions at https://developer.mantidproject.org/Testing/MuonAnalysis_test_guides/index.html
http://developer.mantidproject.org/Testing/MuonInterface/MuonTesting.html

AnthonyLim23, 2021-04-27:
I can load xml in the grouping tab without loading data - no crash but it does produce warnings

the plotting line edits are gone https://github.com/mantidproject/mantid/issues/31234

At a later date we should add a copy values to the sequential fitting -> #31238

On the fitting tab, the checkbox text is cut off #31234

AnthonyLim23, 2021-04-27:
Sequential fitting is not copying the correct log values. All of the fits have the first fit's log values - #31242

AnthonyLim23, 2021-04-27:
When doing the results table for co-added EMU 20918-20 with sample temp the value is not the average -> #31244 this is present in 6.1

AnthonyLim23, 2021-04-27:
still have the ALC and elemental analysis to do

AnthonyLim23, 2021-04-28:
elemental analysis seems ok

AnthonyLim23, 2021-04-28:
just need to test auto-add on the ALC tomorrow

AnthonyLim23, 2021-04-29:
ALC auto-add worked. So everything else looks ok to me

closed 2021-04-29 by AnthonyLim23