mantidproject/mantid #31114: Manual Testing Multi dataset fitting

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: https://www.mantidproject.org/Fitting_QENS_I(Q,_t)

robertapplin, 2021-05-05:
Have there been any issues to come out of this @StephenSmith25 ?

StephenSmith25, 2021-05-05:
Wasn't this a Mantid plot interface? I tested the multi data set fitting you get in the fit browser, but I don't think this one exists right? Or maybe I'm totally wrong

robertapplin, 2021-05-05:
Good point, this is only MantidPlot. @DanielMurphy22 do you think this should be changed to be an issue for testing the fit browser?

DanielMurphy22, 2021-05-05:
I assume this is mostly handled by the ConvFit and I(Q,T) tabs in the Indirect Data Analysis interface now?? I'm tempted to say we just drop this test rather than changing it, assuming the behaviour is tested in the IDA interface?

robertapplin, 2021-05-05:
Yes this type of fitting can be done in Indirect Data Analysis, and I think most of the behaviours is captured in the indirect manual tests. Ok I will remove this from the excel sheet

robertapplin, 2021-05-05:
Thanks for testing the fit browser @StephenSmith25

closed 2021-05-05 by robertapplin