mantidproject/mantid #31116: Manual Testing Engineering Diffraction

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Follow: http://docs.mantidproject.org/nightly/interfaces/Engineering%20Diffraction%20Test%20Guide.html

Pasarus, 2021-04-29:
After entering the vanadium and calibration sample numbers in the initial tab, then I click calibrate, and it pops up a warning box saying Error, Mantid is searching for the files. This isn't very clear to me at least as to why this pops up, as it isn't even searching until I click calibrate for a second time.

Pasarus, 2021-04-29:
Nothing to stop the release :+1: