mantidproject/mantid #31117: Manual Testing Workspace Groups

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Follow the examples at http://docs.mantidproject.org/nightly/concepts/WorkspaceGroup.html 
* Ensure MantidWorkbench displays Group Workspaces correctly in the workspace tree

robertapplin, 2021-04-26:
Sorry, decided to switch you two around!

DannyHindson, 2021-04-29:
Have tested on Windows and it's all fine. There were a couple of minor documentation changes required eg references to EventWorkspace that should be GroupWorkspace. I will create an issue for these

closed 2021-04-29 by DannyHindson