mantidproject/mantid #31119: Manual Testing ISIS Reflectometry

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

For testing procedures see http://developer.mantidproject.org/Testing/ReflectometryGUI/ReflectometryGUITests.html

DavidFair, 2021-05-05:
Things noticed during testing:
- Trying to load OFFSEPC10791 in the sample data throws an exception
- On new batch tabs the runs tab is too small by default, I have to keep dragging it out to see the run numbers
- More of UX feedback, it would be nice if SaveASCII showed the list of logged parameters on single click of a workspace rather than double
Otherwise everything worked as expected

DavidFair, 2021-05-05:
I'll throw an @gemmaguest in so she gets notified ^

gemmaguest, 2021-05-05:
@DavidFair I can't reproduce the problem with OFFSPEC10791. Have you got any more info on this?

I've added a comment on #26539 about the runs column being too small. It's not been highlighted by users but it's hopefully an easy fix and it's really annoying.

Good point, the SaveASCII tab is not the most intuitive. This is somewhat historic though and ideally I'd like to get rid of it altogether, so I'd rather not mess with it for now.

DavidFair, 2021-05-05:
Unfortunately not beyond this is using the ISIS Sample data, so maybe it's corrupt? Full stack trace here:
```
Error reading file C:\Users\ieb35538\Downloads\SampleData-ISIS\OFFSPEC00010791.raw
Error in execution of algorithm LoadNexus:
Unable to read data in File: in "C:\Users\ieb35538\Downloads\SampleData-ISIS\OFFSPEC00010791.raw"
Error in execution of algorithm ReflectometryISISLoadAndProcess:
LoadNexus-v1: Unable to read data in File: in "C:\Users\ieb35538\Downloads\SampleData-ISIS\OFFSPEC00010791.raw"
  at line 81 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/ReflectometryISISLoadAndProcess.py'
  caused by line 214 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/ReflectometryISISLoadAndProcess.py'
  caused by line 352 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/ReflectometryISISLoadAndProcess.py'
  caused by line 1048 in 'C:\MantidNightlyInstall\bin\mantid\simpleapi.py'
```

The run is set to 10791 angle 1 in the table

gemmaguest, 2021-05-05:
Ok I'll try redownloading it in case it's changed. The copy I have is fine

gemmaguest, 2021-05-05:
@DavidFair It looks like the offspec problem is a Windows-only problem with LoadISISNexus. I've raised #31365. Good spot!

closed 2021-05-06 by DavidFair