mantidproject/mantid #31120: Manual Testing ISIS SANS (new GUI)

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

See https://developer.mantidproject.org/Testing/SANSGUI/SANSGUITests.html

DanielMurphy22, 2021-04-13:
### Note for tester
there is a known bug that hasn't yet been fixed: https://github.com/mantidproject/mantid/issues/30306

Matt-Cumber, 2021-04-29:
Testing on Mac OS

Cannot run 2D reduction from manual testing instructions

robertapplin, 2021-05-18:
Reopened for some more testing after some bugs have been fixed

sf1919, 2021-05-18:
I've run through the SANS GUI Testing on Windows 10 and have the following comments:-

- The csv file includes the column name can_sans. By process of elimination I believe that is the Can Scatter column from the GUI. Could it be renamed to make it more obvious?
- I've been having trouble with changing data in the Runs table. It often hangs and when it evenutally sorts itself out the GUI has disappeared. Thankfully it hasn't closed, I think it's just somehow minimised it but it's a bit of a pain
- For the Runs Table Editing part of the testing when I tried to put a number in Slice in the Settings tab I get a pop up box warning. For example when I tried 8 I got ```‘The provided event slice configuration [‘8’] cannot be parsed because of 8’```. I am assuming this is user error but am mentioning it just in case
- I did manage to change some of the settings and then exported the file. When I loaded the file back the settings that I had changed were back to default (I changed Absolute Scale on General, Scale, Event Slice, Sample tab and Scattering Incident monitor to 1 on Adjustment tab). Maybe I chose the wrong settings to change? 
- In Processing 1D reduction it says you should get 6 files per row. I got 4 per row and the text before this suggests it should be 4. A typo in the testing guide?
- Also in Processing 1D reduction not only did the incorrect Sample Scatter turn the row blue and give an error when hovered over it also produced a Mantid error pop up and lots of error text in messages pane
- In Processing Merged reduction instructions it says I should get a plot with 6 lines - I got 8.
- I couldn't do the Beam centre finder test as neither ```Update main-detector``` or ```Update Hab``` chexkboxes appear on the Beam centre tab

Otherwise everything else went fine. 

Just to note - I did not run any of the tests on my 4K display setup. Let me know if you want me to do additional 4k tests.

DavidFair, 2021-05-18:
Thanks Sarah

>     * The csv file includes the column name can_sans. By process of elimination I believe that is the Can Scatter column from the GUI. Could it be renamed to make it more obvious?

Unfortunately not, the CSV files don't have any sort of marker for versions and these changes (AFAIK) would break backwards compatibility. Interestingly enough I think your the first person to notice this (including myself), so it's likely been misleading for years.

>     * I've been having trouble with changing data in the Runs table. It often hangs and when it evenutally sorts itself out the GUI has disappeared. Thankfully it hasn't closed, I think it's just somehow minimised it but it's a bit of a pain

This one is a bit concerning, as you using the ISIS Data archive search? Or is it just local files

> * For the Runs Table Editing part of the testing when I tried to put a number in Slice in the Settings tab I get a pop up box warning. For example when I tried 8 I got `‘The provided event slice [...]

That looks correct, as a slice has a starting to end value in Angstrom. If you try `8-10` does it work?

> * I did manage to change some of the settings and then exported the file. When I loaded the file back the settings that I had changed were back to default (I changed Absolute Scale on General, [...]

This is a known bug that unfortunately didn't make this release #30306 

>     * In Processing 1D reduction it says you should get 6 files per row. I got 4 per row and the text before this suggests it should be 4. A typo in the testing guide?

Did it save the .h5 and .xml? If it didn't save any .nxs then it means it's not saving the unsubtracted can and sample - I'll have a look into that...
 
>     * Also in Processing 1D reduction not only did the incorrect Sample Scatter turn the row blue and give an error when hovered over it also produced a Mantid error pop up and lots of error text in messages pane

Yeah this is normal, we print the full stack trace to the log as sometimes it's useful for bugs and (so far) the noisiness of an error hasn't bothered them. So that sounds correct.

>     * In Processing Merged reduction instructions it says I should get a plot with 6 lines - I got 8.

I'll look into this as well and either update the docs / find where those extra ones came from

>     * I couldn't do the Beam centre finder test as neither `Update main-detector` or `Update Hab` chexkboxes appear on the Beam centre tab

I'm working on the beam centre tab for LARMOR having incorrect scaling at the moment. So I can go through and test that. You'll need to load your data before it works

> Otherwise everything else went fine.
> 
> Just to note - I did not run any of the tests on my 4K display setup. Let me know if you want me to do additional 4k tests.

Great genuinely thanks for testing. 
Regarding scaling, could you simply open it on a 4k display just to get a quick feel that it lays out sensibly. I don't think of our current users have mentioned it though so you're definitely ahead of the curve. Just more to get a feel if we will need to fix this long-term

sf1919, 2021-05-19:
Thanks David - Sounds like you have everything in hand!

I did the testing using my own local files (the training dataset referenced in the testing). So it may well just be my computer not coping. Might be worth someone else giving that particular bit of the test another go.

I will do some 4K testing next. I haven't picked up on anything before (see #31271 ) but its always good to double check.

sf1919, 2021-05-19:
Ok so just a couple of things I've noticed on 4K setup. One is fairly minor the other isn't but if none of the users have had problems so far I reckon they can probably be looked at in a future sprint rather than as part of release.

Minor issue
The main menu on the left hand side of the GUI isn't wide enough to read all the options. However there is a scroll bar so there is a sort of work around.

[image]

Major issue
I know you only asked me to look at the first screen but I'm kind of glad I went through some of the other menus and tabs! In Settings the Q, Wavelength, Detector Limits tab is unusable!
[image]

Not sure how you want to handle these in terms of issue tickets but if you do create tickets to deal with these, don't forget to connect them to the 4K Epic #31271

closed 2021-06-02 by robertapplin