mantidproject/mantid #31122: Manual Testing Mantid Basics Course

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

* Build a docs-html target of Mantid
* Check against the online documentation (http://docs.mantidproject.org/nightly/tutorials/mantid_basic_course/index.html#mantid-basic-course)
* Open up the basic course (docs/html/tutorials/mantid_basic_course/index.html)
* Check that the pages in there make sense

SilkeSchomann, 2021-04-28:
There is still a section for MantidPlot. Do we still need this?

SilkeSchomann, 2021-04-28:
This was tested on Windows both with the current nightly build (2021-04-27) and with a debug build of release-next. After building docs-html I had to build docs-qtassistant and docs-qthelp as well. Otherwise the auto generated plots were not available.

SilkeSchomann, 2021-04-29:
The release build of release-next is fine as well after building the targets described above.

PR #31290 (2021-04-30): Removed Mantid Plot from Mantid basics course  
**Description of work.**

Both the menu entry for Mantid Plot and all files for the respective folder were removed.

**To test:**

Check the tutorial for the Mantid Basics Course. There should be no section for Mantid Plot anymore.

Fixes #31122 . 

*This does not require release notes* because it is only an update of the documentation of the Mantid Basics Course.

---

#### Reviewer ####

Please comment on the following ([full description](http://developer.mantidproject.org/ReviewingAPullRequest.html)):

##### Code Review #####

- Is the code of [... cut]
closed 2021-04-30 by SilkeSchomann