mantidproject/mantid #31125: Manual Testing Error Reporter

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

http://developer.mantidproject.org/Testing/ErrorReporter-ProjectRecovery/ErrorReporterTesting.html

clayton-tom, 2021-04-29:
The email address field should regex the input to make sure it's a valid email address

clayton-tom, 2021-04-29:
Text box chars > 3200 causes a server error, only send first 3200 chars if more

clayton-tom, 2021-04-29:
Entering zero or one characters in the free text box doesn't change the Share non-ii button, entering >= two greys it out and it won't ungrey out - the free text box should be sent if the case that the user chooses to share non-ii

clayton-tom, 2021-04-29:
Opened #31293 #31292

closed 2021-04-29 by clayton-tom