mantidproject/mantid #31126: Manual Testing Sliceviewer

DanielMurphy22, 2021-04-12:
You have been assigned manual testing. The hope is to catch as many problems with the code before release, so it would be great if you can take some time to give a serious test to your assigned area. Thank you!!

The general guide to manual testing:

* The tests must be performed on the installer versions of the final release candidate. Not on local compiled code.
* Serious errors involving loss of functionality, crashes etc. should be raised
as issues with the current release as a milestone and an email sent to the project manager immediately.
* Minor and cosmetic issues should be raised as issues against the forthcoming
releases.
* First try things that should work, then try to break Mantid, e.g. entering invalid values, unexpected characters etc.
* Don't spend more than a few hours on the testing as fatigue will kick in.
* If you find errors in the documentation, please correct them.
* Comment against this ticket the OS environment you are testing against.
* Close the this issue once you are done.

### Specific Notes:

Test for both MDEvent and MDHisto workspaces: Briefly I tested for both MDEvent and MDHisto workspaces:
[ ] 1. Viewing 2D,3D and 4D MD data - (changing bins and sliders, color limits, colormap etc.)
[ ] 2. Panning, home, Zooming/Dynamic rebinning - both by selecting region with mouse and scrolling
[ ] 3. Toggling tracking of cursor
[ ] 4. Toggle gridlines
[ ] 5. Experiment with line plots and box integration (disabled fro non-orthogonal data)
[ ] 6. Changing viewing axes
[ ] 7. Peak overlay (multiple peaks workspaces, integrated peaks, sorting table, zooming on peaks, changing axes with peaks shown, peak overlay should not be shown for 2D data)
[ ] 8. Repeat (1-7) for data with non-orthogonal axes
[ ] 9. Play around with window - resizing, adjusting peak table size
[ ] 10. Saving plot
[ ] 11. Delete/rename underlying workspaces (at present sliceviewer does not update)

RichardWaiteSTFC, 2021-05-05:
Have found some more major issues that may be related to the ADS observer
1. Crash when having selected a peak, changing overlaid peak table and re-clicking a peak #31362 
2. Deleting a peak in workspace deletes it from both non-integrated and integrated peak tables in sliceviewer #31358 

Some issues with non-orthogonal view
1. Spherical shapes not getting transformed #31275 (will be fixed by PR #31288)
2.  Cursor coordinates in ImageInfoWidget not correct for non-orthog view #31273 

The behaviour of the slider value box (where the slice center of the integrated dimensions is specified) is a bit counter-intuitive - when the user enters a value it doesn't rebin the data but jumps to nearest bin center of histo workspace (#31285)

Some more minor usability issues: #31355, #31353

Some things that would be nice to have (but I haven't made issues for)
1. Re-ordering table columns in peak viewer to have Run, H, K, L, I, Sig, I/Sig (i.e. important info) first so easier to see
2. Resize peak overlay pop-up to encompass the whole workspace name

RichardWaiteSTFC, 2021-05-05:
The spectrum viewer-like capability looks OK - there was the issue #31266 regarding panning that has already been fixed

closed 2021-05-05 by RichardWaiteSTFC