mantidproject/mantid #31133: Expose saveImage() from InstrumentViewer to python

DanielMurphy22, 2021-04-13:
**Original reporter:** Davide Raspino / ISIS

This is a request to add in the small feature to save an image of instrument viewer in a script.

Instrument viewer in workbench was exposed to python 6 months ago: https://github.com/mantidproject/mantid/pull/29904
The new docs page shows what is currently possible: https://docs.mantidproject.org/nightly/workbench/instrumentviewer.html?highlight=instrumentview%20python#python-control

[Previously in MantidPlot](https://www.mantidproject.org/MantidPlot:_Instrument_View#Python_control) there was a saveImage method that was rather useful for the user. I think the code to do this is still there in CPP (Maybe InstrumentWidget.cpp), it just hasn't been exposed to Python.

gvardany, 2021-04-15:
This is a requirement for us too. However, we found that it is not as trivial. @gui-co Can you share your findings?

gui-co, 2021-04-15:
I wrote a message on slack about that few weeks ago. I wanted to create a python export of `InstrumentWidgetRenderTab::saveImage`
https://github.com/mantidproject/mantid/blob/33c39c0f4264f323bec0d308b94f5dd3e95343d4/qt/widgets/instrumentview/inc/MantidQtWidgets/InstrumentView/InstrumentWidgetRenderTab.h#L76

The python function was working but the exported image did not contain anything useful. The following warning was also reported : `QOpenGLContext::swapBuffers() called with non-exposed window, behavior is undefined`. I tried to resolve it by waiting that the widget is valid (see https://doc.qt.io/qt-5/qopenglwidget.html#isValid) but it did not help.

Enclosed is a patch that expose the function in python and the exported image produced by the following script:
```python
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
from mantidqt.widgets.instrumentview.api import get_instrumentview
CreateSampleWorkspace(OutputWorkspace="ws")
i = get_instrumentview(mtd["ws"])
i.save_image(<file_path>)
```

<details>
<summary>patch</summary>

```diff
diff --git a/qt/python/mantidqt/widgets/instrumentview/_instrumentview.sip b/qt/python/mantidqt/widgets/instrumentview/_instrumentview.sip
index 0936496cedd..b83f44d620c 100644
--- a/qt/python/mantidqt/widgets/instrumentview/_instrumentview.sip
+++ b/qt/python/mantidqt/widgets/instrumentview/_instrumentview.sip
@@ -300,6 +300,8 @@ public:
     Reset the view widget
 %End
 
+    void saveImage(QString filename);
+
 private:
   // constructors all private
   InstrumentWidgetRenderTab();
diff --git a/qt/python/mantidqt/widgets/instrumentview/presenter.py b/qt/python/mantidqt/widgets/instrumentview/presenter.py
index c4620697f8b..06d24c1e6b9 100644
--- a/qt/python/mantidqt/widgets/instrumentview/presenter.py
+++ b/qt/python/mantidqt/widgets/instrumentview/presenter.py
@@ -53,6 +53,9 @@ class InstrumentViewPresenter(ObservingPresenter):
     def show_view(self):
         self.container.show()
 
+    def save_image(self, filename):
+        self.container.save_image(filename)
+
     def get_current_tab(self):
         return self.container.get_current_tab()
 
diff --git a/qt/python/mantidqt/widgets/instrumentview/view.py b/qt/python/mantidqt/widgets/instrumentview/view.py
index f7c19a4ed69..0cee68a2d52 100644
--- a/qt/python/mantidqt/widgets/instrumentview/view.py
+++ b/qt/python/mantidqt/widgets/instrumentview/view.py
@@ -89,6 +89,10 @@ class InstrumentView(QWidget, ObservingView):
             new_window_name = new_ws_name
         self.widget.replaceWorkspace(new_ws_name, new_window_name)
 
+    def save_image(self, filename):
+        self.show()
+        self.widget.saveImage(filename)
+
     @Slot()
     def _run_close(self):
         self.close()
diff --git a/qt/widgets/instrumentview/src/InstrumentWidget.cpp b/qt/widgets/instrumentview/src/InstrumentWidget.cpp
index 245ce467c61..0148a0d6668 100644
--- a/qt/widgets/instrumentview/src/InstrumentWidget.cpp
+++ b/qt/widgets/instrumentview/src/InstrumentWidget.cpp
@@ -66,6 +66,8 @@
 #include <numeric>
 #include <stdexcept>
 #include <utility>
+#include <chrono>
+#include <thread>
 
 using namespace Mantid::API;
 using namespace Mantid::Geometry;
@@ -873,6 +875,9 @@ void InstrumentWidget::saveImage(QString filename) {
   }
 
   if (isGLEnabled()) {
+    while (!(m_InstrumentDisplay->isValid())) {
+      std::this_thread::sleep_for(std::chrono::microseconds{1});
+    }
     m_InstrumentDisplay->saveToFile(filename);
   } else {
     m_simpleDisplay->saveToFile(filename);
```

</details>

[image]

gui-co, 2021-04-22:
Calling `getRenderTab` did not help. Apparently the solution is to call `show()` from the outside of the `InstrumentView` and wait until the widget is correcly shown. Something like:
```python
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
import time
from mantidqt.widgets.instrumentview.api import get_instrumentview
CreateSampleWorkspace(OutputWorkspace="ws")
i = get_instrumentview(mtd["ws"])
i.show_view()
time.sleep(1)
i.save_image(<file_path>)
```