mantidproject/mantid #31150: FFT bug

AnthonyLim23, 2021-04-16:
If you run the below script, the discrete fourier transform and numpy fourier transform give the same result. However, Mantid's FFT gives a different result and has more noise.

```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

x = np.linspace(0, 0.5, 500)
y = np.sin(40 * 2 * np.pi * x) + 0.5 * np.sin(90 * 2 * np.pi * x)

ws = CreateWorkspace(x,y)

def DFT(x):
    """
    Compute the discrete Fourier Transform of the 1D array x
    :param x: (array)
    """

    N = x.size
    n = np.arange(N)
    k = n.reshape((N, 1))
    e = np.exp(-2j * np.pi * k * n / N)
    return np.dot(e, x)
    

ft = DFT(y)

np_fft = np.fft.fft(y)

mantid = FFT(ws)
FT = CreateWorkspace(mantid.readX(0), ft/np.max(ft))
num = CreateWorkspace(mantid.readX(0), np_fft/np.max(np_fft))

plt.plot(mantid.readX(3), mantid.readY(3)/np.max(mantid.readY(3)), label="Mantid")
plt.plot(FT.readX(0),FT.readY(0), label="discrete FT", linestyle="-")
plt.plot(num.readX(0), num.readY(0), label="numpy FFT", linestyle="--")
plt.xlim(0,120)
plt.legend()
plt.show()
```

AnthonyLim23, 2021-04-16:
[image]

AnthonyLim23, 2021-04-19:
This also looks bad. First open Muon analysis load `62260`. Then run `PaddingAndApodization` on one the asymmetry groups (e.g. MUSR62260; Group; bottom; Asymmetry; MA). Set the apodization function to Gaussian and the output as `test`. Then run the below script

```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
from mantid.kernel import PhysicalConstants as const

ws = AnalysisDataService.retrieve("test")
x = ws.readX(0)
y = ws.readY(0)

def DFT(x):
    """
    Compute the discrete Fourier Transform of the 1D array x
    :param x: (array)
    """

    N = x.size
    n = np.arange(N)
    k = n.reshape((N, 1))
    e = np.exp(-2j * np.pi * k * n / N)
    return np.dot(e, x)
    

ft = DFT(y)

np_fft = np.fft.fft(y)

mantid = FFT(ws,AcceptXRoundingErrors=True)
def convert_to_field(workspace_name):
    """
    Apply the ConvertAxisByFormula algorithm to convert from MHz to Field.
    """
    alg = AlgorithmManager.create("ConvertAxisByFormula")
    alg.initialize()
    alg.setAlwaysStoreInADS(True)
    alg.setProperty("InputWorkspace", workspace_name)
    alg.setProperty("OutputWorkspace", workspace_name)
    alg.setProperty("Formula", 'x * 1. / '+str(const.MuonGyromagneticRatio))
    alg.setProperty("AxisTitle", 'Field')
    alg.setProperty('AxisUnits', 'Gauss')
    alg.execute()
    return alg.getProperty("OutputWorkspace").valueAsStr

convert_to_field("mantid")

FT = CreateWorkspace(mantid.readX(0), ft)
num = CreateWorkspace(mantid.readX(0), np_fft)

def y_data(y):
    norm = np.sqrt(np.max(y)*np.max(y))
    print(norm)
    return y/norm

plt.plot(mantid.readX(3), y_data(mantid.readY(3)), label="Mantid")
plt.plot(FT.readX(0),y_data(FT.readY(0)), label="discrete FT", linestyle="-")
plt.plot(num.readX(0), y_data(num.readY(0)), label="numpy FFT", linestyle="--")
plt.xlim(0,120)
plt.legend()
plt.show()
```

The results look bad
[image]

RichardWaiteSTFC, 2021-04-20:
I have tried this with some simple functions (using Transform=Backward in the mantid FFT) - though I have scaled the mantid and numpy curves for easy comparison (numpy defaults to backward normalisation but I think there's still a factor of N somewhere). Basically the modulus looks OK (compares well with numpy except for the pseudo delta-func - I am still looking into this) and odd functions have a purely imag FFT and even functions have a real FFT as expected. I think there is a sign issue for the imag part of the transform when comparing to numpy: though I thought the FFT of a sin is ~ delta(freq-f0) - delta(freq+f0)  which agrees with what mantid produces.
[image]
I have tried to rep[roduce the Gaussian example in the mantid FFT docs (I don't have the code or the parameters) but it looks OK again (but for sign change of imag part).
[image]
I am not sure if the sign of the imag part matters - most examples seem to only use the modulus of the complex FFT.

RichardWaiteSTFC, 2021-04-20:
As discussed offline, looking at the docs for a forward transform (default in mantid),  an FFT of a real function produces an output workspace with 6 spectra, the modulus of the FFT is the last spectra (index 5), but in your original code is it looks like you are plotting index 3 (which I believe is the real part?). Maybe it's worth plotting index 5 in your second example (I am unfamiliar with the muon analysis tab and couldn't follow your instructions to make the test workspace).

AnthonyLim23, 2021-04-20:
In muons they like to look at both the real and imaginary parts. So I was not looking at the modulus. It is also possible that the modulus is ok because we have the right number of bugs in the real and imaginary parts, that they just cancel out (like the 2 minus sign mistake).

AnthonyLim23, 2021-04-20:
For the other example

1. Open muon analysis
2. Load MUSR 62260 (you can close the GUI)
3. Open the PaddingAndApodization algorithm dialog
4. Set the input to be one of the groups (e.g. MUSR62260; Group; bottom; Asymmetry; MA)
5. Set the function to gaussian and output to test
6. Run the algorithm
7. Run the script

RichardWaiteSTFC, 2021-04-20:
> In muons they like to look at both the real and imaginary parts. So I was not looking at the modulus. It is also possible that the modulus is ok because we have the right number of bugs in the real and imaginary parts, that they just cancel out (like the 2 minus sign mistake).

OK interesting, well I don't see a problem with the real component from any of the examples above (e.g. even funcs, odd funcs etc.)  but as you say mistakes could somehow cancel to give the right modulus so it looks like we need to inspect the imag and real parts in more detail. I will think about other examples we can use to quantitatively test the algorithm is working.

AnthonyLim23, 2021-04-20:
It should be possible. For example
F(x) = 3x + 6*i*x -> mod(F(fx)) = sqrt(9+36)
but swapping the real and imaginary parts will give the same modulus
G(x) = 6x + 3ix

AnthonyLim23, 2021-04-20:
Here is Richards script:
```
# import mantid algorithms, numpy and matplotlib
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np
def doMantidFFT(x,y):
    ws = CreateWorkspace(x,y)
    wsFFT = FFT(InputWorkspace='ws', OutputWorkspace='wsFFT', Transform="Backward")
    scale = max(wsFFT.readY(2))
    return wsFFT.readX(2), wsFFT.readY(2)/scale, \
        wsFFT.readY(0)/scale, wsFFT.readY(1)/scale 
def doNumpyFFT(x,y):
    np_fft = np.fft.fft(y)
    np_freq = np.fft.fftfreq(x.size, d=x[1]-x[0])
    scale = max(abs(np_fft))
    return np_freq, abs(np_fft)/scale, \
        np.real(np_fft)/scale, np.imag(np_fft)/scale
x = np.linspace(-5, 5, 201)
# y = np.ones(x.shape) # Constant signal
# y = np.zeros(x.shape); y[10] = 2 # delta-func (ish!)
# y = np.sin(2*2*np.pi*x) # sine - odd fun so FFT is imag
# y = np.cos(1*2*np.pi*x) + np.cos(2*2*np.pi*x) # 2 cos (f and 2f) - even FFT is real!
# y = np.exp(-2*(x-1)**2) # Gauss
# y = np.sin(1*2*np.pi*x) + np.sin(2*2*np.pi*x) # sin(f) + sin(2f)
# numpy FFT
np_freq, np_fft, np_fft_real, np_fft_imag = doNumpyFFT(x,y)
# mantid FFT
mtd_freq, mtd_fft, mtd_fft_real, mtd_fft_imag = doMantidFFT(x,y)
fig, ax = plt.subplots(4,1)
# signal
ax[0].plot(x, y, '-ok', label='signal')
ax[0].legend()
ax[0].set_ylabel('Signal')
# magnitude
ax[1].plot(np_freq, np_fft, '-ok', label='numpy')
ax[1].plot(mtd_freq, mtd_fft, '--r', label='mantid')
ax[1].plot(mtd_freq, np.sqrt(mtd_fft_real**2 + mtd_fft_imag**2), '.b', label='mantid sqrt(im^2 + re^2)')
ax[1].legend()
ax[1].set_xlabel('freq')
ax[1].set_ylabel('FFT amp')
# real
ax[2].plot(np_freq, np_fft_real, '-ok', label='numpy')
ax[2].plot(mtd_freq, mtd_fft_real, '--r', label='mantid')
ax[2].legend()
ax[2].set_xlabel('freq')
ax[2].set_ylabel('FFT real')
# imag
ax[3].plot(np_freq, np_fft_imag, '-ok', label='numpy')
ax[3].plot(mtd_freq, mtd_fft_imag, '--r', label='mantid')
ax[3].legend()
ax[3].set_xlabel('freq')
ax[3].set_ylabel('FFT imag')
fig.show()
[axx.set_xlim(-5,5) for axx in ax[1:]]
```

If you use `y = np.sin(1*2*np.pi*x) + np.sin(2.5*2*np.pi*x) # sin(f) + sin(2f)` on line 27, then you get an odd set of peaks. They point in opposite directions and should both point the same way.

RichardWaiteSTFC, 2021-04-20:
> If you use `y = np.sin(1*2*np.pi*x) + np.sin(2.5*2*np.pi*x) # sin(f) + sin(2f)` on line 27, then you get an odd set of peaks. They point in opposite directions and should both point the same way.
[image]

I've been thinking about this - when you FFT a sine wave you get a positive delta func on the negative freq. side and a negative one on positive freq. side (or vice-versa I can't remember). Since `FFT{f + g} = FFT{f} + FFT{g} ` you expect e.g. the delta func on the positive side to have the same sign - which is what we see in mantid but not what numpy produces so maybe I'm wrong?

AnthonyLim23, 2021-04-20:
What I am seeing is the other way around to the labels on your plot. 
```
from mantid.simpleapi import *
import matplotlib.pyplot as plt
import numpy as np

x = np.linspace(0, 0.5, 500)
y = np.sin(40 * 2 * np.pi * x) + 0.5 * np.sin(90 * 2 * np.pi * x)

ws = CreateWorkspace(x,y)

def doMantidFFT(x,y):
    ws = CreateWorkspace(x,y)
    wsFFT = FFT(InputWorkspace='ws', OutputWorkspace='wsFFT', Transform="Backward")
    return wsFFT.readX(2), wsFFT.readY(2), wsFFT.readY(0), wsFFT.readY(1) 

mtd_freq, mtd_fft, mtd_fft_real, mtd_fft_imag = doMantidFFT(x,y)

def y_data(y):
    norm = np.sqrt(np.max(y)*np.max(y))
    print(norm)
    return y/norm

def doNumpyFFT(x,y):
    np_fft = np.fft.fft(y)
    np_freq = np.fft.fftfreq(x.size, d=x[1]-x[0])
    return np_freq, abs(np_fft), np.real(np_fft), np.imag(np_fft)

np_freq, np_fft, np_fft_real, np_fft_imag = doNumpyFFT(x,y)

plt.plot(mtd_freq, y_data(mtd_fft_imag), label="Mantid")
plt.plot(np_freq, y_data(np_fft_imag), label="numpy FFT", linestyle="--")
plt.xlim(0,120)
plt.legend()
plt.show()
```

AnthonyLim23, 2021-05-05:
I have had another look at this.

For the forward transform, setting `AutoShift=True` improves the results

For backwards it seems to make no difference

AnthonyLim23, 2021-05-05:
Also a round trip does not seem to be possible, unless I did something stupid. @RichardWaiteSTFC what do you think?

```

x = np.linspace(-5, 5, 201)
# y = np.ones(x.shape) # Constant signal
# y = np.zeros(x.shape); y[10] = 2 # delta-func (ish!)
# y = np.sin(2*2*np.pi*x) # sine - odd fun so FFT is imag
# y = np.cos(1*2*np.pi*x) + np.cos(2*2*np.pi*x) # 2 cos (f and 2f) - even FFT is real!
# y = np.exp(-2*(x-1)**2) # Gauss
y = np.sin(1*2*np.pi*x) + np.sin(2*2*np.pi*x) # sin(f) + sin(2f)
ws = CreateWorkspace(x,y)

mantid = FFT(ws,AcceptXRoundingErrors=True, Transform="Forward")

time = FFT(mantid,AcceptXRoundingErrors=True, Transform="Backward")
```

AnthonyLim23, 2021-05-06:
I got the round trip to work, I missed some arguments.

AnthonyLim23, 2021-05-06:
apparently if we are changing this it is worth switching to FFTW as it is faster and GSL even recommend it

AnthonyLim23, 2021-05-06:
changing this line https://github.com/mantidproject/mantid/blob/f65f29f21dc9bb5ad80e457e73a606418be68696/Framework/Algorithms/src/FFT.cpp#L196 to `double im1 = -re * s + im * c;` seems to help. As far as I can tell it only rotates the few examples we had with strange rotations. 

Long term, I think its worth considering a switch the FFTW

PR #31458 (2021-05-13): Frequency domain analysis FFT default values
**Description of work.**
The think that the reason for the complaints about FFT's was because the default values were not helpful. 

I have made the FFT pure real by default and hidden the auto shift option. This is technically a regression in functionality. However, I do not think the shift was being set manually by users and I have not found an example where you would want to set it manually. I have just hidden it for now, so we can put it back in if the users want it.

**To test:**

Open frequency domain analysis
Load EMU 113083
Go to the transfo [... cut]
closed 2021-05-18 by AnthonyLim23