mantidproject/mantid #31196: SANS - Reduced Slice Object missing attribute

DavidFair, 2021-04-22:
The following reduction results in the following error

Sample SANS 65485, sample TRANS 65488, can SANS 65484, can TRANS 65487.
The user file is 202N on Sans2d.
Report to SANS/Najet 
```

No spectra affected.

No spectra affected.

Error in execution of algorithm SANSSingleReduction:

'ReducedSlice' object has no attribute 'data_type'

  at line 108 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/SANS/SANSSingleReduction.py'

  caused by line 184 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/SANS\SANSSingleReductionBase.py'

  caused by line 260 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/SANS/SANSSingleReduction.py'

  File "C:\MantidNightlyInstall\scripts\SANS\sans\gui_logic\models\batch_process_runner.py", line 80, in _process_states_on_thread

    self.batch_processor([state.all_states], use_optimizations, output_mode, plot_results, output_graph, save_can)

  File "C:\MantidNightlyInstall\scripts\SANS\sans\sans_batch.py", line 33, in __call__

    return self._execute(states, use_optimizations, output_mode, plot_results, output_graph, save_can=save_can)

  File "C:\MantidNightlyInstall\scripts\SANS\sans\sans_batch.py", line 42, in _execute

    single_reduction_for_batch(state, use_optimizations, output_mode, plot_results, output_graph,

  File "C:\MantidNightlyInstall\scripts\SANS\sans\algorithm_detail\batch_execution.py", line 125, in single_reduction_for_batch

    reduction_alg.execute()

 

'ReducedSlice' object has no attribute 'data_type'

  at line 108 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/SANS/SANSSingleReduction.py'

  caused by line 184 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/SANS\SANSSingleReductionBase.py'

  caused by line 260 in 'C:/MantidNightlyInstall/plugins/python\algorithms/WorkflowAlgorithms/SANS/SANSSingleReduction.py'

```

smk78, 2021-04-23:
You'll also need the Direct beam: 65487

smk78, 2021-04-23:
This worked fine for me on my office NDW (W10/x64):
[image]

smk78, 2021-04-23:
I had a Zoom call with Najet and got her to demonstrate the issue to me.

She had downloaded and installed a more recent Nightly and it was still playing up. I verified her Manage Directories settings (which were reading from, and writing to, NDLSANSD4) and even had her remove those settings to read from \\isis\inst$\NDXSANS2D data and mask folders and write to C:\temp (what I had done above). But this made no difference at all.

Mantid would not load the files either on Process or on Load; it just threw an error (but not the Mantid error handler).

DavidFair, 2021-04-23:
I'm unable to reproduce this either locally: 
[image]

Could we check the following with Najet, as it's the only machine reproducing the problem currently:
- Najet is using the same numbers for the sample / can direct (as they aren't provided)
- Uninstall Mantid Nightly, then delete `C:\MantidNightlyInstall` as any scripts with changes (such as even a newline being entered by mistake) will not be updated by the installer, then reinstall 
- That the save options aren't a factor (I'm testing with memory only)
- Lastly, check another machine. If all but Najet's machine run this correctly it's likely a local issue. I'm happy to work through it with her to resolve it, but at least it lets us know this won't hold up the release

DavidFair, 2021-06-08:
I'm going to close this as resolved unless we see it come up again

closed 2021-06-08 by DavidFair