mantidproject/mantid #31197: SANS Index out of range for processed data

DavidFair, 2021-04-22:
Report to SANS/Sarah

The batch is attached and I’m using user_sans2d_202I_2p4_4m_M4_Growney_12mm_Changer.txt
In settings I change the wavelength range to RangeLin 2:1:16
Batch file in email provided (GH is playing up for me so I can't upload it at the moment
 

```
  File "/opt/mantidnightly/scripts/SANS/sans/gui_logic/models/batch_process_runner.py", line 80, in _process_states_on_thread

    self.batch_processor([state.all_states], use_optimizations, output_mode, plot_results, output_graph, save_can)

  File "/opt/mantidnightly/scripts/SANS/sans/sans_batch.py", line 33, in __call__

    return self._execute(states, use_optimizations, output_mode, plot_results, output_graph, save_can=save_can)

  File "/opt/mantidnightly/scripts/SANS/sans/sans_batch.py", line 43, in _execute

    save_can=save_can)

  File "/opt/mantidnightly/scripts/SANS/sans/algorithm_detail/batch_execution.py", line 173, in single_reduction_for_batch

    save_to_file(reduction_packages, save_can, additional_run_numbers, event_slice_optimisation=event_slice_optimisation)

  File "/opt/mantidnightly/scripts/SANS/sans/algorithm_detail/batch_execution.py", line 1135, in save_to_file

    transmission_can = to_save[2][i] if to_save[2] else ''

 

list index out of range
```

smk78, 2021-04-22:
The batch file:
[Batch_csv.txt](https://github.com/mantidproject/mantid/files/6359818/Batch_csv.txt)

smk78, 2021-04-22:
I've tried this batch file on a W10/x64 machine using Mantid Nightly 6.0.20210421.2323. Asked it to process every other line with wavelength binning 2:1:16 step 0.05 and all worked fine.

DavidFair, 2021-04-22:
Thanks Steve, I did say in my reply that I'm sure I had fixed this error once before - could we get Sarah to confirm that the issue is resolved in the latest nightly before we close?

I wonder if it's related to the output type, e.g. memory only, vs saving to file or both?

smk78, 2021-04-22:
I've tried memory only and saving everything to all three file types. Didn't get Sarah's error.
But it might be worth asking her if she was doing something else.

DavidFair, 2021-04-23:
On follow up it turns out this is related to processing more n files. The first one will run correctly but all following ones will produce said error

DavidFair, 2021-04-26:
Could not reproduce on Windows / Linux / IDaaS. I've contacted the user to confirm and possibly do a video call as we might be doing some different steps

DavidFair, 2021-04-26:
This missing puzzle piece was setting `RangeLin` and `2:1:16` we can now reproduce this

smk78, 2021-04-26:
Just re-run earlier test in 6.0.20210424.659 on W10/x64. Still cannot reproduce Sarah's error.
[image]
[image]

DavidFair, 2021-04-26:
Is the save mode set to both? This was the other key piece which I had still set from my earlier testing

I can reproduce it on my machine so it should be resolved in the coming day(s) at least

smk78, 2021-04-26:
No it wasn't. But it is now... and lo and behold:
```
GroupWorkspaces started (child)
GroupWorkspaces successful, Duration 0.00 seconds
  File "C:\MantidNightlyInstall\scripts\SANS\sans\gui_logic\models\batch_process_runner.py", line 80, in _process_states_on_thread
    self.batch_processor([state.all_states], use_optimizations, output_mode, plot_results, output_graph, save_can)
  File "C:\MantidNightlyInstall\scripts\SANS\sans\sans_batch.py", line 33, in __call__
    return self._execute(states, use_optimizations, output_mode, plot_results, output_graph, save_can=save_can)
  File "C:\MantidNightlyInstall\scripts\SANS\sans\sans_batch.py", line 42, in _execute
    single_reduction_for_batch(state, use_optimizations, output_mode, plot_results, output_graph,
  File "C:\MantidNightlyInstall\scripts\SANS\sans\algorithm_detail\batch_execution.py", line 173, in single_reduction_for_batch
    save_to_file(reduction_packages, save_can, additional_run_numbers, event_slice_optimisation=event_slice_optimisation)
  File "C:\MantidNightlyInstall\scripts\SANS\sans\algorithm_detail\batch_execution.py", line 1135, in save_to_file
    transmission_can = to_save[2][i] if to_save[2] else ''

list index out of range
```
Happy days!

martyngigg, 2021-05-10:
Fixed by #31246