mantidproject/mantid #31214: Sliceviewer Cursor widget not properly sized for MD workspaces

DanielMurphy22, 2021-04-26:
**Original reporter:** [On the forum](https://forum.mantidproject.org/t/mantid-sliceviewer-cursor-position-unreadable/729)

### Expected behavior
The widget has enough space to display the x,y and signal values. On _my_ Windows machine, this works well.
[image]

### Actual behavior
On _my_ Ubuntu and MacOS machines I found the problem and as you can see the values aren't visibnle as the widget isn't correctly sized. The user found this problem on Windows, so this isn't OS specific, but should be tested on all OS!
[image]
[image]

### Steps to reproduce the behavior
Run this script to create an MD workspace:
```
ws=CreateMDHistoWorkspace(Dimensionality=2,Extents='-3,3,-10,10', \
SignalInput=range(0,100),ErrorInput=range(0,100),\
NumberOfBins='10,10',Names='Dim1,Dim2',Units='MomentumTransfer,EnergyTransfer')
```
Right-click on the workspace and show sliceviewer. Check if the widget is sized correctly when you hover the cursor over the image.
### Platforms affected
All

DanielMurphy22, 2021-04-28:
Of slight relevance to this is how different sliceviewer can look on different screen resolutions. The images included on this testing are a good example: https://github.com/mantidproject/mantid/issues/31111#issuecomment-827733637

sf1919, 2021-05-05:
I have checked and following the steps to reproduce the behaviour I can confirm that on a lower resolution display on Windows 10 I do not have a problem but if I switch to 4k display then I get the same problems seen on Mac and Ubuntu.

closed 2021-05-10 by sf1919