mantidproject/mantid #31225: AlignAndFocusPowder contains deprecated algorithms: MaskInstrument + AlignDetectors

marshallmcdonnell, 2021-04-26:
### Expected behavior
Use `AlignAndFocusPowder` w/o error warnings related deprecated functions

### Actual behavior
Getting current errors when using `AlignAndFocusPowder`
```
MaskInstrument-[Error] MaskInstrument is deprecated (on 2020-07-16). Use MaskDetectors instead.
AlignDetectors-[Error] AlignDetectors is deprecated (on 2021-01-04). Use ConvertUnits instead.
```

*Mantid Version*: 6.0.20210422.2105 (currently nightly at time of writing issue)
*Package*: [conda mantid framework](https://anaconda.org/mantid/mantid-framework)

### Steps to reproduce the behavior

Running `AlignAndFocusPowder` using example and example files from [docs page](https://docs.mantidproject.org/nightly/algorithms/AlignAndFocusPowder-v1.html) should give the following output:

```
FrameworkManager-[Notice] Welcome to Mantid 6.0.20210422.2105
FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
Load-[Notice] Load started
LogParser-[Warning] Cannot process ICPevent log. Period 1 assumed for all data.
InstrumentDefinitionParser-[Notice] Geometry cache is not available
InstrumentDefinitionParser-[Notice] Creating cache in /root/.mantid/instrument/geometryCache/POWGEN243edfd42ebf5d0140e5f79fd93cf4f644e0815f.vtp
Load-[Notice] Load successful, Duration 1.42 seconds
AlignAndFocusPowder-[Notice] AlignAndFocusPowder started
AlignAndFocusPowder-[Notice] Could not find property manager
LoadCalFile-[Warning] 20676 errors (invalid Detector ID's) found when reading .cal file 'pg3_mantid_det.cal'.
MaskInstrument-[Error] MaskInstrument is deprecated (on 2020-07-16). Use MaskDetectors instead.
AlignDetectors-[Error] AlignDetectors is deprecated (on 2021-01-04). Use ConvertUnits instead.
MaskDetectors-[Notice] Masking using workspace indicies
MaskDetectors-[Warning] No spectra affected.
AlignAndFocusPowder-[Notice] AlignAndFocusPowder successful, Duration 1.35 seconds
Finished!!!
```

To get the above output, I personally used the following setup w/ Docker + Conda package to run the test python script that reproduces the error:

*Dockerfile*
```
FROM continuumio/miniconda3:4.9.2
  
ARG MANTID_VERSION="6.0.20210422.2105"

# Get POWGEN test files
RUN wget -O pg3_mantid_det.cal https://testdata.mantidproject.org/ftp/external-data/MD5/e2b281817b76eadbc26a0a2617477e97 \
    && wget -O PG3_9830_event.nxs https://github.com/mantidproject/systemtests/blob/master/Data/PG3_9830_event.nxs?raw=true

RUN conda create -n env python=3.7
RUN echo "source activate env" > ~/.bashrc
ENV PATH /opt/conda/envs/env/bin:$PATH

# Install mantid framework
RUN conda config --add channels conda-forge --add channels mantid/label/nightly \
    && conda install -n env mantid-framework=="${MANTID_VERSION}"

ADD test.py .
```

*test.py*
```
from mantid.simpleapi import AlignAndFocusPowder, Load
  
PG3_9830_event = Load('PG3_9830_event.nxs')
PG3_9830_event = AlignAndFocusPowder(
    PG3_9830_event,
    CalFileName='pg3_mantid_det.cal',
    Params='100')

print("Finished!!!")
```

*docker commands*
```
docker build -t mantid-conda .
docker run mantid-conda python test.py
```

martyngigg, 2021-08-23:
I'm rolling this over by default. Please pull it back to 6.2 if it's required.

martyngigg, 2022-01-06:
I'll roll this again. Please pull it back if necessary.

peterfpeterson, 2023-01-24:
`MaskDetectors` is unused anywhere in all of mantid.

`AlignDetectors` is mentioned in several algorithms' "`seeAlso()` method

closed 2023-01-24 by peterfpeterson