mantidproject/mantid #31234: muon plotting ranges

AnthonyLim23, 2021-04-27:
On nightly I do not have the x or y boxes (beneath the plot) for selecting the plotting range. 

[image]

Matt-Cumber, 2021-04-27:
This is strange, looks okay on Mac
[image]

AnthonyLim23, 2021-04-27:
could you look on windows?

Matt-Cumber, 2021-04-27:
Still looks okay to me, could be a display size thing

AnthonyLim23, 2021-04-27:
@sf1919 how does yours look?

sf1919, 2021-04-27:
Which GUI are you using?

Matt-Cumber, 2021-04-27:
Muon Analysis or FDA

sf1919, 2021-04-27:
Ok I have tried both Muon Analysis and FDA on both my laptop and my other screen and all of them show the same as they do for Anthony. In fact the Autoscale y and Errors appear slightly cropped.

[image]

FYI I'm running Windows 10 on 4K laptop and extra screen.

AnthonyLim23, 2021-04-27:
@sf1919 and @Matt-Cumber could you two work together on this to make sure it works on Windows and mac (we can make a mac installer)? I will test it on Linux during the review.

AnthonyLim23, 2021-04-27:
Im updating this to cover general display problems as I have found another one on the fitting tab - the text is truncated 

[image]

martyngigg, 2021-05-04:
Fixed by #31311

PR #31311 (2021-05-04): Fix display formatting issues on Muon plotting and Fitting tabs for 4k displays
**Description of work.**
This fixes display issues caused by using Muon Analysis and Frequency Domain Analysis with a 4k display.

**To test:**
1. Open Mantid
2. Open Interfaces >> Muon >> Frequency Domain Analysis
3. Check that X max, X min, Y max and Y min appears beneath the plotting window
4. Open the Fitting Tab 
5. Check that Plot Guess, Selected Workspace and Fit Status on the left hand side of the screen can be read in full.
6. Close The Frequency Domain Analysis GUI
7. Open Interfaces >> Muon >> Muon Analy [... cut]
closed 2021-05-04 by martyngigg