mantidproject/mantid #31235: Multiple Scattering tidy up

DannyHindson, 2021-04-27:
Fix a few miscellaneous issues that Spencer Howells has raised relating to the CalculateMultipleScattering algorithm:

Documentation:
1) category "CorrectionFunctions\Absorption Corrections" is incorrect - should be simply CorrectionFunctions or else new subcategory of multiple scattering
2) references section at the bottom isn't rendering correctly on Ubuntu - when there are back references present the colour of the reference number at the start of the reference gets set to White so v hard to see
3) the type of the NumberScatterings parameter is "m" on Ubuntu. This is because the type of the parameter is set to size_t in the code which isn't mapped to a friendly name in Framework\Kernel\src\Property.cpp. Will set type to int like all the others so it is described as "Number" in the documentation. It's a bit of a pain supplying literal values into a size_t property because C++ doesn't have a literal type for size_t

Plus general functionality:
1) the summed workspace is wrong. It should be the sum of 2 to n not 1 to n
2) workspace containing ratio J<sub>1</sub>/&Sigma; J<sub>i</sub> (as done in Discus) is missing. This is used to apply the multiple scattering correction via a "ratio" method as opposed to subtracting the Sum

DannyHindson, 2021-04-27:
Re documentation point 2), the problem occurs when there are multiple references to a single footnote in the main body of the text. Sphinx\docutils inserts back references to these in the reference. These are rendered OK on Windows:

[image]

...but not on Ubuntu

I tried setting the docutils footnote_backlinks setting to false in a docutils.conf file but this didn't fix it - in fact it made it worse because all the reference numbers were white even if a reference is mentioned only once in the main text. Will just refer to the reference a single time in the main text as a workaround

martyngigg, 2021-05-10:
Fixed by #31281

closed 2021-05-10 by martyngigg