mantidproject/mantid #31238: Copy values in sequential fitting

AnthonyLim23, 2021-04-27:
In sequential fitting we may have lots of runs (30+) and it is annoying to manually set them all. We should add a way to set the values once and then copy it to the others. 

Possible methods:
- Have a `copy selected parameters to all` button
- Have a `copy selected parameters to all` in a context menu

AnthonyLim23, 2021-05-18:
@robertapplin and @sf1919 can you do this one together?

robertapplin, 2021-07-08:
I think the best method would be to use a context menu in order to keep the interface neat. So the workflow would be to right click on a table cell, and then to select `Set to All` to get the behaviour we want. We can implement this in several steps:

- [ ] 1. Create a context menu that appears when you right click a cell in the `SequentialTableWidget`
https://stackoverflow.com/questions/6094452/qtablewidget-context-menu-only-for-specific-cells-columns
- [ ] 2. Connect the `Set to All` menu item that you create to a slot in the `seq_fitting_tab_presenter`
- [ ] 3. From the slot, get the name of the selected parameter, and its new value from the selected cell
- [ ] 4. Set the value of this parameter for all the domains in the `tf_asymmetry_fitting_model`
- [ ] 5. Update the `seq_fitting_tab_view` based on the updated information in the model

AnthonyLim23, 2021-07-08:
it could just be the general fitting model

robertapplin, 2021-07-08:
After discussion with @sf1919 , we decided that a radio button would be better than a context menu. When the radio button is on, a change to a parameters value is copied to all domains. When the radio button is off, the parameters value is only changed for the single selected domain.

closed 2021-07-26 by martyngigg