mantidproject/mantid #31240: Monitors not being excluded from limits in sliceviewer when home pressed

RichardWaiteSTFC, 2021-04-27:
When viewing data that includes monitors that has been converted from TOF to d-spacing, the initial x-axis limit in the sliceviewer appears to be set OK (this should have been fixed in PR #30365), but when you press home the x limits are reset to be the TOF limits. This problem only occurs with data that includes monitors (which cannot be converted from TOF to d-spacing) - it appears they are not being excluded when the limits are calculated when the home button is pressed.

Found by: Pas (WISH)

RichardWaiteSTFC, 2021-04-28:
The inital limits and extent are set correctly in `samplingimage.py` (same code used for colorfill plots which do not have this bug) but the actual problem arises in the sliceviewer `model.py `(L195) in the function `get_dim_limits()`. The workaround in samplingimage.py (that loops over all spectra) isn't present - it just calls `getDimension(0).getMaximum() `etc. which aren't working correctly - see this example code
```
w = Load(Filename=r'SXD23767.raw', OutputWorkspace='SXD23767')
ConvertUnits(InputWorkspace='SXD23767', OutputWorkspace='SXD23767', Target='dSpacing')
w = mtd[‘SXD23767’]
dimx = w.getDimension(0)
dimx.getMinimum() # 0.21 – should be 0.1865
dimx.getMaximum() # 5.11 – should be 35.89!
```

martyngigg, 2021-04-30:
Fixed by #31254

PR #31254 (2021-04-29): Get home limits from full extent of SamplingImage if sliceviewer plotting a matrix workspace
**Description of work.**

When plotting a ragged matrix workspace the extent of the data is determined by looping over all spectra (excl. monitors) and getting the min and max x-value. However when the home button is pressed in sliceviewer the limits were recalculated in the model based on `ws.getDimension(xindex).getMinimum()` and  `.getMaximum()` - which is determined from the extent of the first spectrum (if this is a monitor then this will get the TOF extent rather than e.g. the d-spacing so can be [... cut]
closed 2021-04-30 by martyngigg