mantidproject/mantid #31244: Results table should give average for co-add

AnthonyLim23, 2021-04-27:
Load EMU 20918-20
Click co-Add
Go to fitting and do a fit
Go the results table tab and tick `sample temp`
Press the `Output Results`
Look at the table and the value is 200, it should be the average `(180+190+200)/3`

Matt-Cumber, 2021-04-27:
Related to #30254

AnthonyLim23, 2021-06-22:
the sample logs for the co-added data is wrong. So the problem is at the loading stage with the plus alg

```

a = Load("EMU20918")
b = Load("EMU20919")

c = a+b

for ws in [a,b,c]:
    run = ws[0].run()
    print(run.getLogData('sample_temp').value)
```

so we will need to manually average the sample logs when combining the data

https://github.com/mantidproject/mantid/blob/master/scripts/Muon/GUI/Common/utilities/load_utils.py#L276

sf1919, 2021-06-29:
I've been looking into this and MergeRuns could be a good solution. It looks like it uses the Instrument Parameter File for the default merging behaviour for sample logs. Otherwise you can add to or override the defaults to:-

- Sum
- Combine as a time series
- Combine as a list
- Rebin if the bins are different

As well as allowing to set warnings/failure notifications as relevant.

However it does not look like we can override and average anything. Could this be something we set in the Instrument Parameter file & would we want to?

If not then we'd need to look at using MergeRuns with some additional code to handle anything that falls outside of the extras MergeRuns currently offers. It still seems like a better solution than writing a completely new piece of code that is potentially duplicating MergeRuns.

closed 2024-05-22 by github-actions[bot]