mantidproject/mantid #31247: Indirect FQFit - Crash when loading result data

robertapplin, 2021-04-28:
### Expected behavior
The data should be loaded ok.

### Actual behavior
The interface hangs and eventually crashes.

### Steps to reproduce the behavior
1. Indirect Data Analysis interface
2. Got to FQFit tab
3. Load the data below.
4. Hangs and then crash
[irs26176_graphite002_red_Conv_seq_2L_0-50__Result.zip](https://github.com/mantidproject/mantid/files/6390047/irs26176_graphite002_red_Conv_seq_2L_0-50__Result.zip)

### Platforms affected
Found on windows, probably all

SpencerHowells, 2021-04-28:
@robertapplin It needs result from ConvFit with delta function. So file should have eg *_Delta1L*

SpencerHowells, 2021-04-28:
@robertapplin The Fit function menu should also be input type specific. Single Input is ok.

robertapplin, 2021-04-28:
> @robertapplin It needs result from ConvFit with delta function. So file should have eg __Delta1L_

I see, I'll try find a way to stop the crash for non-Delta result data

robertapplin, 2021-04-28:
> @robertapplin The Fit function menu should also be input type specific. Single Input is ok.

I have looked at the fit function menu and I have found that it is showing the correct functions for each parameter type. Is this not working for you only for Multiple Input?

SpencerHowells, 2021-04-28:
@robertapplin Yes. It is only wrong for Multiple. Single is fine.

robertapplin, 2021-04-28:
> @robertapplin Yes. It is only wrong for Multiple. Single is fine.

For reference this now has its own issue #31251

martyngigg, 2021-05-05:
Fixed by #31252

PR #31252 (2021-05-04): Indirect FQFit - Fix crash when loading data that has no EISF
**Description of work.**
This PR fixes a crash in FQFit when loading data that has no EISF parameter.

**To test:**
1. Indirect Data Analysis interface
2. Go to FQFit tab
3. Load the data below. It might take a while to load, but it should not crash.
4. Choose any fit function and click `Run`
5. You should see a fit in the plot when it is finished fitting.
[irs26176_graphite002_red_Conv_seq_2L_0-50__Result.zip](https://github.com/mantidproject/mantid/files/6390047/irs26176_graphite002_red_Conv_seq_2L_0-50__Result.zip)

Fixes #31247
 [... cut]
closed 2021-05-05 by martyngigg