mantidproject/mantid #31248: Indirect Symmetrise - Cannot change Emin to be negative

robertapplin, 2021-04-28:
### Expected behavior
The Emin should be allowed to be negative

### Actual behavior
Dragging the slider or changing EMin in the table to be negative doesn't seem to be allowed anymore.

### Steps to reproduce the behavior
1. Indirect->Data Reduction-> Symmetrise tab
2. Follow the example workflow in https://docs.mantidproject.org/nightly/interfaces/indirect/Indirect%20Data%20Reduction.html#symmetrise-example-workflow

EMin doesn't seem to be allowed to be negative

### Platforms affected
Found on windows, probably all

SpencerHowells, 2021-04-28:
@robertapplin This routine was requested but hardly, if ever, used. It is not particularly user-friendly - even I struggle to run it. The input has positive energy range much bigger than negative range. The idea is to take the extra positive range & reflect it on the negative side. The emax slider is used to choose the reflecting point & I think emin is linked to it. Emax is then moved so that the y value for emax & emin are the same.

SpencerHowells, 2021-04-28:
@robertapplin Try this file I think I've worked it out. Emin & Emax define the positive range to be reflected. Move Emin to be half way up the positive side of the peak & you will see 2 horizontal lines. These are the y values for +Emin & -Emin. The Emin line should then be adjusted so that the re & blue merge (or near enough). Then click on preview & you will see the result. So I think it's all ok.
[osi_red.zip](https://github.com/mantidproject/mantid/files/6391885/osi_red.zip)

martyngigg, 2021-05-10:
Fixed by #31359

PR #31359 (2021-05-06): Indirect - Fix Symmetrise range selectors not moving below zero
**Description of work.**
This PR fixes a bug in the Symmetrise tab of Indirect Data Reduction where the range selectors could not be moved below zero. This was caused by the bounds not having been set for the range selectors. These bounds default to be between 0 and 1, and so the markers could not be moved as intended.

No tests added because this can't be tested easily.

**To test:**
1. Indirect->Data Reduction-> Symmetrise tab
2. You should be able to follow the example workflow in https://docs.mantidproject.org/nightly/interfac [... cut]
closed 2021-05-10 by martyngigg